grbl/grbl #724: Error: Unsupported command

beckmx, 2015-06-03:
Hello guys, I was trying to do a simple shape and noticed it was doing a bad movement, it then started to reset X,Y positions and well it broke all the milling it was doing, the code it was trying to execute is this one:

G0 X192.95862944162437 Y52.79746192893401
G1 Z-0.5 F95
G1 X170.251269035533 Y52.796954314720814 F110
G3 X169.18781725888326 Y52.58629441624365 I0 J-2.7766497461928936
G3 X168.28680203045684 Y51.984771573604064 I1.0634517766497462 J-2.565989847715736
G3 X167.68274111675126 Y51.08121827411168 I1.9644670050761421 J-1.9644670050761421
G3 X167.47208121827413 Y50.017766497461935 I2.5685279187817263 J-1.0634517766497462
G3 X167.68274111675126 Y48.95685279187817 I2.7741116751269033 J0
G3 X168.28680203045684 Y48.05583756345178 I2.565989847715736 J1.0634517766497462
G3 X169.18781725888326 Y47.45177664974619 I1.9644670050761421 J1.9644670050761421
G3 X170.251269035533 Y47.24111675126903 I1.0634517766497462 J2.5685279187817263
G1 X192.9593908629442 Y47.24111675126903

The code was generated from the parka website. and I am using version 0.9i, any ideas?

ashelly, 2015-06-03:
Grbl has a line limit of 80 characters. Several of these lines exceed it.
I doubt you need all that precision, you could probably round to 4 or 5
decimal places and be more accurate than your machine.

chamnit, 2015-06-03:
@beckmx : @ashelly is right. There is an 80-ish character limit per line in Grbl, but you should have gotten a `Line overflow` error to let know something went wrong. Do you not get these?

beckmx, 2015-06-03:
I was using UniversalGcodeSender and it splits the decimals down to 4, but
I have been seen errors with the GCODE I have, but I don't know why, if you
want to, I can share it with you, it is an enclosure with many cuts for a
raspberry

chamnit, 2015-06-03:
@beckmx : What are the errors exactly that you are getting. That will help identify the cause.

beckmx, 2015-06-03:
In the console I only detected the error "T0 M6" it says, it doesn't
support it. But beyond from that, I don't see any errors in the console.
But what I do see are errors in the movement of the machine, It is milling
right and then it makes little pauses and kinda loses its zero coordinates
and keeps milling the right pattern in the wrong place.

chamnit, 2015-06-03:
@beckmx : T0 is not supported. See the supported g-code list in the Wiki.

Intermittent movement is indicative of an electrical problem, which is probably why you also had a problem with your g-code stream. It's probably corrupting the USB data.

closed 2015-07-02 by chamnit