grbl/grbl #725: F word doesn't match G20/G21

109JB, 2015-06-03:
I noticed that when $13=1 and G21 is active, issuing a F word does a conversion from inches to mm.  For example, if a F25.4 is issued, a $G shows the F value at 1.0.

Likewise, if $13=0 and G20 is active it does a conversion in the other direction. For example if F1 is issued, the $G shows F=25.

Also, changing the display units ($13) changes the value of the  F word as above.

This is not how it should be. Regardless of the displayed units, the parser values should be in the units corresponding to G20/G21, whichever is active at the time. If there is a change in G20/G21 then the F value should change as well, but it should not change when the display units are changed ($13).

It appears that GRBL is using the display units ($13) as the units that the user is inputting the f-word as.

chamnit, 2015-06-03:
@109JB : Grbl tracks everything internally as mm. When F is printed in the $G state report, it looks at the $13 setting to determine which units to print back to the user. I suppose you are right, especially with inverse time (not sure what Grbl prints out for this now that I think about it). The F value should show the programmed value, rather than it's interpreted value, but this could lead to some confusion as well.

However, it's really simple to make this change. That is if no one any problems with it or has a dissenting opinion or issue with this.

109JB, 2015-06-04:
Just an FYI on this. I checked my linuxCNC install, and the Axis gui for LinuxCNC has a box that continually reports modal values including the F value.  It reports the value according to G20/G21, whichever is active regardless of the units displayed on the DRO. Changing the units display on the DRO does not affect what is reported in the modal report.

matthijskooijman, 2015-12-20:
I guess this is just a matter of documenting a behaviour and sticking to it. It might make sense to use the same units for displaying F-values as are used for inputing new values (which is the proposal here), but one could also argue that reporting F-values should use the same unit as used when reporting positions, which is exactly what $13 is meant for, AFAIK. If this is changed, it is something to do for 1.0 I guess, due to backward compatibility of GUIs.