grbl/grbl #726: Problem with v0.9i

Petrovski, 2015-06-04:
Hello,

I have been testing GRBL since a year. I never used it on a machine, but I did try out some GUIs and did spin some motors. 

When I did a pull from the repo to get the latest changes and imported grbl in Arduino IDE, I got a 

    Grbl 0.9i ['$' for help]
    ['$H'|'$X' to unlock]
    
When I then issue a "?" command, I get

    <Alarm,MPos:214748.3648,214748.3648,214748.

I do then reset using

    $X

and either the serial terminal dies (does not respond anymore) or I get a 

    [Caution: Unlocked] 

response. I then perform a query again

    ?

and receive

    <Idle,MPos:214748.3648,130485.2096,214748.                                                                 
    Grbl 0.9i ['$' for help]
    ['$H'|'$X' to unlock]

Anybody knows what is wrong?

My setup is:

1. Change the default cpu_map to *CPU_MAP_ATMEGA2560*
2. Change the target cpu in Makefile to *atmega2560*
3. Copy grbl-master/grbl to ~/sketchbook/libraries/
4. Start Arduino IDE, compile and upload grbl example

109JB, 2015-06-05:
your machine positions are HUGE. 214748.3648 inches, or 17895.697 feet or 3.389 miles.  I will guarantee you don't have a machine with that much travel.

Do a hard reset. This will reset the machine position to zero. Then check the motor settings for reality ($100-$132).

Petrovski, 2015-06-05:
Hello,

I reinstalled Arduino IDE from 1.1.xx to 1.6.4 and all is fine now.

Best Regards

chamnit, 2015-06-05:
@Petrovski : Thanks for reporting the fix. That's strange that the version change altered things, but I'll keep this in mind if someone else comes along with a similar problem.

109JB, 2015-06-05:
I personally don't think it was the IDE version change that fixed it. I actually tested it on my GRBL install that has been running perfectly. I changed the stepper setting so it wouldn't take forever, but I ran the machine position to what was shown in the original post (214748.3648 inches) and got the exact same results.. A simple hard reset took operation back to normal.

Petrovski, 2015-06-05:
Hello,

@109JB, it was a fresh install and all that could have been reset was reset. Must have to do with incompatibility of my version of the OS with Arduino 1.1.xx or vice versa.

ashelly, 2015-06-05:
@109JB,  214748.3648 is (2^31) / 10000.  It almost certainly not a real position, but a sign of memory corruption somewhere.

Petrovski, 2015-06-15:
@ashelly Quote: "214748.3648 is (2^31) / 10000." ***This is impressive***. How could you know this? :) Alright, anyway, I got the same problem again. Did some motor testing with my 328p and pristine grbl v.9i. It went well until I increased the pulse frequency, perhaps, beyond the capabilities of the chip. I think It was something like I used UniversalGcodeSender for testing purposes.

chamnit, 2015-06-15:
@Petrovski : You are exceeding Grbl's 30 kHz step/sec capability. This limit is the same on the the mega2560. You may be able ok to exceed this limit a little without much issue, as I've put a safety margin on this limit. Grbl doesn't do any step/sec checks at the moment, since it takes up valuable code space to do so. Please refrain from exceeding this limit because problems like this one will come up.

closed 2015-06-15 by chamnit

Petrovski, 2015-06-15:
OK, sure I will refrain from using in production, but refrain from testing is against my belief system :) The limits were clearly exceeded, but what could have been damaged, so that I can't re-upload a fresh grbl and start testing again? Could it be that the atmega328 got fried? Or some issue with my PC?

Any ideas?

chamnit, 2015-06-15:
@Petrovski : Not sure what exactly could cause you not being able to upload Grbl, but something like this did happen to me once. I had to re-flash the boot loader onto my 328p because it got corrupted for some reason. It's possible that it happened to you as well. If you don't have an ISP programmer, you can use another Arduino to act as one. There is documentation on Arduino's website on how to do this.

Petrovski, 2015-06-15:
@chamnit : I do have an AVRISPmkII and did re-flash but no success. I upload arduino examples and they seem to work, even those with Serial port interaction. However, when I recompile grbl and upload it I get the same errors as described in the first post above.

Will report if I find out what's going on.

Thanks!

ashelly, 2015-06-15:
@Petrovski, after a few years of embedded systems programming, you will also be able to recognize powers of two on sight. :)  The /10000 probably comes from the `N_DECIMAL_COORDVALUE_INCH  = 4` setting in the coordinate print routine.      
I have occasionally had issues with garbled eeprom settings after using avrdude to upload a new image to the mega.  One of the first things I do after an upload is use `$$` to check the settings.

Petrovski, 2015-06-16:
Hi all,

Thinking about the potential pulse frequency overload, I was running at about ~67kHz prior the breakdown or:

    ( 8000 (mm/min) / 60 (sec/min) ) * 500 steps/mm =  66.6 steps/sec or 66.6kHz

@chamonit: Could this not be checked at every $100 and $110 set/save and to print out a warning/error message that the max pulse frequency is exceeded?

chamnit, 2015-06-16:
It's there but not enabled. It's not often people grossly exceed this limit so I opted to save a little flash space for now.