grbl/grbl #727: Homing Limit Question

dsteury, 2015-06-08:
All functions seem to be working perfectly as far as Hard Limits are concerned. I have inverted the pins ($5=1) and use NC switches. The machine will effectively recognize all limits and stop accordingly. My problem is that Homing does not seem to work. It will travel for about 10mm on the z-axis (My first "homing axis") but then stops, goes into "Alarm" mode and gives the error "Alarm: Homing fail." As it has not hit any limits nor has it exceeded soft limits (I have turned them off for now) I cannot see what the issue might be. Is there an option to use hard limits as homing switches and I just don't have it selected?Any suggestions would be greatly appreciated.

chamnit, 2015-06-08:
This means that the homing cycle skipped the search phase of homing. The 10mm move is part of the locate cycle, after the search phase. This is likely due to a problem in the limit switch wiring or electrical noise. With NC switches, the NC switch must be connected to ground such that the pin reads 0V low during normal operation.

dsteury, 2015-06-08:
That makes sense, I was wondering if some part of the process was being completed. I am confused however why grbl will allow movement immediately prior to homing. I do not receive false limit tripping during a set of code either. Every code/file I put in runs flawlessly and I have tested all limits repeatedly. The switches are all connected on the low side of a 1k resistor that connects to ground when the switches are closed. (Normal operation since all switches are NC) The other side of the resistor is connected to 5v. I cannot see an error in the wiring is there any way to test what exactly is going wrong? (i.e. a more specific error code) Thank you for the quick reply by the way. I am very grateful for the assistance.

chamnit, 2015-06-08:
@dsteury : You can enable an undocumented status report that will show limit switch states in the '?' query. Set `$10=255` to enable all of the status reports. You'll see a `Lim:000` or similar. 0 is low and 1 is high in the order ZYX. Change $10 back to the previous value when done. Some GUIs don't understand some of these new status reports.

dsteury, 2015-06-12:
Ok I finally got a few minutes to try this. I have changed "$10=255" with no issues. However, when I type "$?" I get the typical menu. 
(Options: $$, $#, $G, $I, $N, $x, $Nx, $C, $X, $H, ~, ?)
If I type "$?" again I get the same menu and I cannot seem to locate the "Lim:000" for switch status. I also noticed that when I invert the direction of the ports ($3=5) it does not change the homing direction ($23) so I set them both to 5 so the homing will occur in the negative direction I have my motors set to. This led me to again think that homing might not be able to work with limit switches inverted or might need a separate command so that homing switches are inverted as well? Thanks again for all the help.

dsteury, 2015-06-12:
I should mention the programs I tried to access these menus in were ChiliPeppr and UniversalGcodeSender on Mac OSx

chamnit, 2015-06-12:
Try using a serial terminal and only send '?', not with a $. GUIs tend to pick off the ? character and hide the status reports.

dsteury, 2015-06-12:
OK I tried the command at resting state with no alarms and got this

<Idle,MPos:0.000,0.000,6.000,WPos:0.000,0.000,1.000,Buf:0,RX:2,Lim:100>

dsteury, 2015-06-12:
*edit
<idle,MPos:0.000,0.000,6.000,WPos:0.000,0.000,1.000,Buf:0,RX:2,Lim:100>

Sorry this text box is not paste friendly

dsteury, 2015-06-12:
edit

"<Idle,MPos:0.000,0.000,6.000,WPos:0.000,0.000,1.000,Buf:0,RX:2,Lim:100>"

dsteury, 2015-06-12:
edit
 Idle,MPos:0.000,0.000,6.000,WPos:0.000,0.000,1.000,Buf:0,RX:2,Lim:100

dsteury, 2015-06-12:
So the first bit in the limit mask is the Z limit?

chamnit, 2015-06-12:
Yes. Lim: is in the order ZYX. 0 is low and 1 is high. If you have the invert limit pin disabled (NO switch to GND), this means your X and Y axes are interpreted as triggered. If you have invert limit pin enabled (NC switch to GND), this means your Z axis is interpreted as triggered. Either way, it looks like your limits are mis-wired.

dsteury, 2015-06-12:
Perfect I have NC switches so I will troubleshoot the Z-axis switch. One last question, in 0.9i how do you determine if spindle pwm is enabled? I want to ensure my Z-axis limit switch is going to the correct pin. I have it currently going to the fourth pin up (i.e. Stepper Enable/Disable, X-Limit, Y-Limit, Z-Limit) Just want to make sure it shouldn't be "Stepper Enable/Disable, X-Limit, Y-Limit, Variable Spindle PWM, Z-Limit"

closed 2015-07-02 by chamnit