grbl/grbl #728: HOMING_FORCE_SET_ORIGIN issue

pjamestx, 2015-06-08:
I have recently upgraded my X-Carve to a newer build of grbl 0.9i.  I understand that after homing, it places the workspace entirely in the negative quadrant, which is the behavior I was seeing after the upgrade.

I wanted to get back to the behavior where it would zero out the axes after homing.  I followed the instructions in the FAQ section and uncommented the HOMING_FORCE_SET_ORIGIN line in my config.h file and recompiled/uploaded.  After homing, I'm still seeing non-zero values in Chilipeppr.  Previously they were ~-790, the value of my max travel settings, but now I'm seeing X/Y/Z values of -21.320/-28.940/+17.510.

As a test, I went back and incremented the N_HOMING_LOCATE_CYCLE value to 2, to make sure that I was compiling/uploading correctly, and I confirmed that it changed from doing a single homing cycle per axis to two cycles.

The only two differences in my config.h file are:

#define N_HOMING_LOCATE_CYCLE 2 // Integer (1-128)
and
#define HOMING_FORCE_SET_ORIGIN // Uncomment to enable.

Below are my current grbl settings:
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=1 (step port invert mask:00000001)
$3=3 (dir port invert mask:00000011)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.050 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=750.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=188.976 (z, step/mm)
$110=8000.000 (x max rate, mm/min)
$111=8000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=500.000 (x accel, mm/sec^2)
$121=500.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=790.000 (x max travel, mm)
$131=790.000 (y max travel, mm)
$132=100.000 (z max travel, mm)

Has anyone else experienced this behavior?

chamnit, 2015-06-08:
Are you looking at machine coordinates or work coordinates? You may have G54 offsets stored and active. Check with a `$#` command. You should be seeing machine coordinates as [0,0,0] with the most current build.

pjamestx, 2015-06-08:
You are correct, it was a set of work coordinates that had been saved.  Once I opened UGS instead of Chilipeppr it became obvious.  I was able to reset the coordinates using this command:

G10 L2 P1 X0Y0Z0

Thanks for your quick response, and a great product :)

closed 2015-06-08 by pjamestx