grbl/grbl #729: Control & limit pins not on same port ?

maaadbob, 2015-06-09:
>...All CONTROLs pins must be on the same port and not on a port with other input pins (limits).

Why is this ?
Why do the control pins (inputs) need to be on a different port to the limit (input) pins ?

Cheers,
Bob.

chamnit, 2015-06-09:
@maaadbob : Not sure why it matters that they have to share a port, but it's because the pin change interrupt triggers per port. It keeps the limit pin and the control pin ISR functions separate and independent.

closed 2015-06-09 by chamnit

J-Dunn, 2015-07-01:
This can be got around by testing which pins have caused the interrupt when entering the ISR. It's a couple of lines of extra code and a little slower. This 'must be' is done to keep the code as tight and fast as possible. ie it's a 'must be' unless you change the code. 

I'm currently trying to put Z_STEP on a separate port. It means creating a separate constants for Z DDR and Z_DIRPORT etc. but it's doable and does not seem too costly.