grbl/grbl #730: Location of code to plan move in XYZ space ?

maaadbob, 2015-06-09:
Hi all,

Have 4th axis running just fine on my machine (circular), but can only operate three axis at a time.
So can move XYZ or XYC or YZC etc.
Can someone point me to the code section that handles the cartesian planning.

Cheers,
Bob.

chamnit, 2015-06-09:
@maaadbob : Grbl doesn't support 4th axis. Planning is done in the planner code.

closed 2015-06-09 by chamnit

maaadbob, 2015-06-09:
You mean to write, GRBL doesn't support concurrent motion on four axis ?
It can support a 4th axis as my machine is running with a 4th.  :)

Br,
Bob.

chamnit, 2015-06-09:
@maaadbob : No. I meant that Grbl doesn't support 4th axis. Never has officially. Other forks/users have add this. It will soon though.

mkeyno, 2016-04-13:
@maaadbob  what if we use the X axis for A axis to simulate X&Y planar to A&Y cylindrical space

brakthehun, 2019-09-11:
How soon will this be supported. I just need XY and C axis. The "Spindle" pin will run then plunge knife solenoid.
I'd like C to be automatically calculated so I just have to give it XY coordinates and it point the knife the right direction.