grbl/grbl #732: Incorrect status after Feed Hold

bensuffolk, 2015-06-12:
I believe this is a bug, but it might be as intended, if so please let me know :-)

I came across this when implementing jogging with a switch that givens me -1, 0 1. My implementation moves to the machine extremes or -1 or 1, and on 0 issues a feed stop ! and then checks the status of the machine by polling with the ? to see when it's is in "Hold". Once it is, I issue a Soft Reset to clear the rest of the command.

That all sounds like a good plan (if there was a clear butter command then I would not need the reset of course, but I understand thats a different discussion).

However I am getting the status "Hold" back before the machine actually finished stopping. (i.e is still in the "Run" state). When I see that "Hold" status my app is issuing the reset, which in turn causes an alarm as the reset happened during machine movement, and means I could loose position.

So I don't think it should return "Hold" until the machine is actually in a safe Hold state, but it seems to return "Hold" when its entering the "Hold" state.

chamnit, 2015-06-12:
@bensuffolk : This isn't a bug. It was an intentional change to the HOLD behavior in v0.9i. It's also known that this change broke some implementations of continuous jogging with a move, feed hold, and then soft-resetting, like you have done. That said, this isn't the first time this issue has been brought up, so I'll think about a solution for the next version.

closed 2015-06-12 by chamnit

bensuffolk, 2015-06-12:
@chamnit fair enough, so long as you know about it and its not a bug.

Out if interest what the thinking behind doing this? Would another status of "Holding" for the interim phase between "Run" and "Hold" be a simple solution maybe?

ecncshop, 2015-09-07:
Cant we simply aborted a move that has been put on HOLD ( using ! cmd ) ? Say I dont want to RESUME ( ~ cmd) the move. I simply want to Abort the move and start a new move. Only way I can figure out for this is a SoftwareReset ( Ctl X cmd). Is there any other command for this or is there plan to add one in future ?

ecncshop, 2015-09-07:
Problem with SoftwareReset is it compltetely reboots arduino and that takes 1 or 2 seconds , which is not acceptable for my situation. There shud be some other command to abort a move and start a fresh move ?!

chamnit, 2015-09-07:
@ecncshop : This behavior is intentional. The only reason to abort a motion and continue a different one is for jogging. This would never happen in a g-code program, because it'll more than likely corrupt the program. Hence the strict behavior. There are other methods to do jogging as the interfacing to Grbl wiki states. 

The main reason people have requested what you have is when you are doing a continuous jog and need to stop, but you've inputted too many motions into the queue. A continuous jog feature is the slated for v1.0, which should solve this.