grbl/grbl #733: Planner description

vlachoudis, 2015-06-13:
I am trying to write a python routine for the job time prediction of a gcode running with grbl, to include in my sender. I would like to understand a bit more how internally the planner works.

* When a cut is changing direction how much it will be decelerated before and accelerated after the change? Is it too crude to approximate the deceleration for each axis independent from the initial velocity to the one of the next segment?

* How does the junction deviation comes into the game? Every line change is assumed to be connected with a junction arc defined by $11? Is the speed in the juction-arc constant and how it is calculated?

* How grbl will deal if for example the X and Y axis have a different acceleration?

Thanks in advance
Vasilis

chamnit, 2015-06-13:
@vlachoudis : Grbl's planner solves the mathematically optimal motion plan for minimal time based on current running conditions and motions queued in the planner buffer. While probable to get an accurate estimation of running time, it's impossible to get a 100% accurate one, especially when the toolpath has small line segments and there are feed holds.

vlachoudis, 2015-06-13:
Thank you @chamnit

closed 2015-06-13 by vlachoudis