grbl/grbl #734: Compiling GRBL in Arduino or XLoader

joelaw419, 2015-06-14:
Trying to follow the instructions on GitHub to re-compile the Grbl 9i files and find that it compiles and uploads but I cannot tell which version is on the arduino to make sure. I downloaded hex files from github Grbl/Grbl page listed about 1/2 way down. I also found older versions 9g and 8c in hex format.

My problem is that if I compare the v9g and the v9i hex files, they are exactly the same? If I compile the code downloaded from Github (grbl-master) and try to compile using their instructions, the Hex file I get is about 8k smaller and doesn't work.

Been at this 2 days now and am still confused with the results. Why can't I compile v9i and get the same v9i Hex file as the one downloaded? 8k is alot of difference. Am I compiling this correctly? I am using Arduino version 1.6.4 and have the grbl loaded into the Library. It compiles with a notice about low memory available but that is expected since there is only 457 bytes for local variables.

Don't know what I am missing or what to do next.

Thanks,

Joe L.

chamnit, 2015-06-15:
You can get a build date when you send Grbl a $i command. The most recent build is 20150529. Not sure why you are having a hard time. Try removing any previously installed Grbl library and don't alter the example file.

joelaw419, 2015-06-15:
Thanks Chamnit,

I think I have solved the problem?  I had multiple Arduino Libraries and was trying to delete the wrong version.  I could not understand if I deleted GRBL from the library that it was still in the Arduino libraries folder when I open the IDE.  I thought that It was coming from somewhere but could not find it.

Now that I know where the correct version is stored, I can delete and re-install the GRBL library correctly and it seems to work OK.

I am trying to get a laser setup on Shapeoko3 and can't get the post processor in Bobcad or CamBam to agree with what I need to control the laser using PWM on v0.9i.  I wanted to verify that I could compile and install valid code before changing anything and re-compiling.  I need the Spindle to output 0-5v using S0-S255 so now I will try to modify if possible.  I think it is set to 0-1000 now.

I see that others have modified GRBL to have a $L1 (Z ais off) and $L0 (Z axis on) and allows the Z axis to control the laser intensity.  That looks like the easiest way but the GRBL-LaserMod version does not seem to work properly for many other things like settings, $i etc.

I will keep plundering on.

Joe L.

closed 2015-07-02 by chamnit