grbl/grbl #735: Pin13 as "enable" with PWM during bootloader is a problem

lkarlslund, 2015-06-15:
I am currently implementing GRBL on an Arduino Nano 5V 16MHz for a used 80W laser engraver I just bought (with defective controller).

The plan is to skip Z-axis completely, by taking it out of the homing cycle.

Also the laser can to variable power, but sending PWM to a pin on the laser PSU. It has also both active low enable and active high enable pins. I am planning on using the active high enable pin, so if a wire gets disconnected, at least it fails by turning off the laser.

Using "use direction as enable" I can get both.

But ...

When the arduini bootloader starts, it signals success and waits for firmware upload, meanwhile BLINKING the LED on pin13. The PSU I have assumes that no PWM on power and enable high = FULL POWER.

So the laser gets toggled 6-8 times rapidly during boot, which is a disaster waiting to happen.

I know this is not a GRBL problem, rather a bootloader problem. But the choice of pins is a problem.

Suggestions are welcome - moving the enable to another pin would solve it, and I can hack the source ot do so. But I'd rather get an official solution, so others can benefit too.

chamnit, 2015-06-15:
@lkarlslund : AFAIK, D13 can only be used as an output because of that LED. It's also a known problem with the boot loader causing pins to be in an indeterminant state when booting. Burning Grbl with through the ISP header seems to solve most of these issues. 

FWIW, spindle direction was selected for D13. So it was the correct choice for this LED problem that you have brought up, if indeed it is a problem. The pin 13 as spindle enable option is new and was installed as a stop gap for people who wanted a separate spindle enable pin with PWM. It seems that I can't please everyone.

It's possible to move D13's functionality to another pin. It's easy to do. You'll have to add some port initialization lines to the code for the analog port, if moved to A4, and may need to alter some of the spindle enable code to account for the separate port/pin as well.

langwadt, 2015-06-16:
since you don't  need Z why not use Z direction as enable? should also be compatible with standard CAM tools, one step down = laser on, one step up = laser off

lkarlslund, 2015-06-22:
I have solved this by just using the PWM output as pulse-enable for the CO2 laser PSU. Analog/PWM in on the PSU is held high by 5V. This works, and I avoid the nasty enable-while-booting problem.

I know that everything is a compromise, and it's not a bad one you made. I just think that selecting pin 13 as enable-something on this kind of machinery is ASKING for trouble out in the real world.

Thanks for GRBL - I got a defective laser machine cheap, and got it working with $10,- in parts off eBay. I couldn't be happier! ;)

chamnit, 2015-07-02:
@lkarlslund : I'm closing this issue, since there are no other output pins that I can put into pin 13 that won't have this potential issue and pin 13 can only be used as an output due to that LED. IMO, using the spindle PWM as an enable when 0V is read is already a solution. I'll place a disclaimer in the config.h file to make people aware of this potential issue. 

FWIW, I'm still pretty sure that if you flash Grbl directly with a programmer and overwrite the boot loader, so that that is no boot loader, this problem shouldn't exist. You can use the Arduino as a ISP and upload with a programmer option. If you have time, please test this theory.

closed 2015-07-02 by chamnit

stevegt, 2018-07-05:
Even though this issue is closed I'm leaving this comment here for googlers in case it might help others:  

In our local setup (an 80W CO2 laser) we're using m8/m9 (coolant on/off) in the Laserweb settings for the beginning and end of  the generated gcode, and have the laser enable connected to the A3 (coolant) pin.  The large coolant chiller is just hooked to the master switch.

More discussion in #1413.