grbl/grbl #736: Grbl on Arduino mega for Relay Selection

lucio507986, 2015-06-15:
Good day. I didn't knew where to write this so i'll just write it here and offer my apologies if it is out of place.

I'm building a CNC punching press; that is a xy table, a relay-activated die selector, and a relay-activated release mechanism for the press. Since I need 6 dies, that means that I can't use the arduino UNO because there aren't enough pins. I read that one can flash GRBL into an Arduino Mega giving me plenty of pins for the relay activation. But I'll need to run something like M6 and T1-6 for the activation of the pins in the Gcode. Is this possible? How do I assign the tools for each pin in GRBL? If not possible, do you know where to dwell in changing the code?

Thanks a lot

chamnit, 2015-06-15:
@lucio507986 : Yes you can use the Mega 2560 build for this application, but it's not officially supported, so there may be bugs. (Although I haven't heard of any.) You'll need to add g-codes to the parser for your application in gcode.c. They'll need to call custom functions that you'll need to write to enable and disable GPIO pins. It's fairly simple for the latter. Look at coolant_control.c and spindle_control.c for examples.

lucio507986, 2015-06-15:
Thank you so much, I'll dig into it

maaadbob, 2015-06-16:
Hi Lucio,

I believe my company can help you out.
We recently helped a company by producing a modified version of grbl and a supporting custom PCB for them.
Feel free to contact me via info at southern-assemblies dot Co dot uk to see if we can create a solution for you.

Best regards,
Bob.

Connected by MOTOBLUR™

https://github.com/grbl/grbl/issues/736

chamnit, 2015-06-16:
@maaadbob : Please refrain from soliciting for commercial work on this site. Thanks.

maaadbob, 2015-06-16:
@chamnit : Apologies, I didn't intend to cause offence.

gecrby, 2016-03-06:
https://github.com/grbl/grbl/issues/865