grbl/grbl #738: Some thinking: how to use less pin on the Arduino:

onekk, 2015-06-17:
I see that GRBL is becoming a very popular project and more people are demanding improvements, but Arduino UNO is limited in pin count, How about to use "one" pin for many buttons, using the ADC and reading the resistance of the pin, I think almost 10 or 16 buttons, so the resistors value are spaced by some amount to not cause problem with wiring and so on.
Seeing that A0, A1 and A2 are Analog pins so it would be technically feasible, maybe for implementing jogging functionality using maybe A2 and feed/hold and cycle-start/resume on the pin A1 leaving A0 for the E-stop/reset.

Only my two cents.

Many thanks for your work and best regards.

Carlo D.

chamnit, 2015-06-17:
@onekk : Internally, tracking a simple pin change (hardware level) requires virtually no additional code for these input control pins. Just a concise ISR routine. If there are levels and such combined into one pin, the code required will become much much larger and you'd have to do some kind of software debouncing to reliably read the pin. 

Pins aren't the only constraint on the Uno. It's also remaining flash space and available memory. It's pretty maxed out. But don't worry I'm working on a more capable version.

closed 2015-06-17 by chamnit

109JB, 2015-06-17:
@chamnit Is there any thought of abandoning the UNO platform in favor of something like the MEGA or DUE? I have a MEGA now and have not found any issues with GRBL on it.  I realize the DUE doesn't have the EEPROM, so would be problematic to switch to that one, but the MEGA with more pins, more memory and still having EEPROM seems like it would be great from an expandability standpoint and ease of porting to the MEGA. I understand that one of the major concepts of GRBL is the low cost of the board, but even the MEGA can be had in clone form for less than $10, or in genuine Arduino form for just $15 more than an UNO. It just seems that even though GRBL has been able to stretch the usability of the UNO platforms, that there is a day coming when an UNO would be totally maxed out.

chamnit, 2015-06-17:
@109JB : Yes. I've been working hard recently with the next Grbl project. However, Grbl on the UNO will stay, because it's perfectly capable as is for the majority of people. It's also a very very cheap and easy to try. I'll still cram in as much as I possibly can. I won't go into more detail on what I'm doing for the next project, but I'll just say that it'll be a great improvement across the board.

mschorer, 2015-06-18:
that sounds interesting ... i've been doing a bit of research for a next gen platform myself.
so far the TIVA launchpad with tm4c123 is my favourite - i've managed to port (my version of) grbl to it. i hope to do tests with real hardware/motors this weekend.
@chamnit: is it going to be a completely new software, a branch of or rolled into the existing code? i'm experimenting with a hardware abstraction layer via the c-preprocessor: this way code size wouldn't have to increase for the original uno version.
i'll also do a shield/boosterpack for the tiva.
if anybody is interested, code is available here on github, draft schematics on my blog ...

chamnit, 2015-06-18:
@mschorer : With the increased flash space and memory, I'll no longer have to make compromises. It'll essentially be a complete re-write and implementing old ideas I've held on to for years (and a lot of new ones).

Thanks for sharing the Tiva port. I've been evaluating them, but can't figure out why these nice TI Tiva C boards haven't had as much uptake as things like the Teensy or even the Due. For the money, they are incredible. Not sure if it's licensing related, limitations on the board(ROM libraries)/IDE, or something else.

faultylee, 2016-05-09:
@chamnit : I just took a MOOC which use the Tiva C for labs. Coming from PIC and Arduino background, I do find that Tiva C is targeting more advance users. There's more steps involved to get it started, which might be a turn off for absolute beginners. And I guess there's no point for TI to compete on the same level as their competitors. Also, TI might have to retain some revenue for Keil, IAR and the like.

I bet GRBL will perform very well on Tiva C boards.

chamnit, 2016-05-09:
@faultylee : Yes I would agree that Grbl's source code would perform very well on a Tiva C (or anything faster than the 328p). However, I think what most people misunderstand is that you won't get that much more out of Grbl, as its written. You'll definitely get faster max step rates and faster reporting, but the machine itself won't really run significantly better. The mathematical models that Grbl is based on need to take advantage of the addition CPU speed, which is something I've been working on and off of for the last year or so. Getting closer and the critical hurdles have been solved.