grbl/grbl #740: Buffers and character counting streaming protocol issue

109JB, 2015-06-17:
I have finally had a chance to do some more work on my GUI and I think I have come across an issue as it relates to the buffers and the character counting protocol. I have implemented character counting on my GUI and it works well. As you know, this protocol keeps both the planner buffer and the RX buffer pretty full. However, in the case where a line sent to GRBL has an error of some kind, GRBL responds with an error, but by the time the GUI sees the error response, there are more commands in the buffer. What happens is that GRBL issues the error response, but then moves on to the next command and keeps moving. Even if the GUI receives the error and stops sending, there can be many lines already in the buffer.  I perceive this as a pretty big issue for the character counting protocol. I am not sure that it is an issue with the send/respond protocol, but I guess it could be if GRBL does the same with the remaining commands in the planner buffer.  I think the error would be detected before it got to the planner buffer though so wouldn't be a issue there. Since GRBL doesn't know about the streaming protocol used, what I think should happen is that if GRBL senses an error, it should clear any remaining commands in the buffer, and wait for a short time before accepting more commands, or something along those lines. This would prevent commands in the buffer from causing a crash, and the time wait would give the GUI a chance to react and account for communication lag.

chamnit, 2015-06-17:
@109JB : Funny enough, I was just discussing this issue with someone offline. You right though, buffered data will still execute after the error occurs. Perhaps as a stop gap solution, I should force Grbl into a hold state and wait for user interaction, but this could be annoying for people used to ignoring the errors for things like M6 commands. This would retain the commands in the buffer, so a GUI doesn't have to resend it. 

With a forced pause, you basically have two options: Either reset or continue. I suppose it'd be nice to have a clear serial buffer command for this event. So you can have a third option: Re-send corrected command and continue stream.

However, I do recommend that you have your GUI routinely check the g-code program by streaming it to Grbl when it's in $C check mode beforehand. This will help you identify potential problems beforehand and prevent this issue from happening.

109JB, 2015-06-17:
Am I correct that this should not affect the send/response streaming?  I'm thinking it wouldn't since only one command is sent at a time, waiting for a GRBL response. As I understand the inner workings of GRBL, it would receive the command and parse it, checking for errors before putting it in the planner buffer and sending either an "ok" or "error" response. Since the GUI is waiting for some kind of response before sending another command it would be caught there so would not be an issue in send/response. Only valid commands are sent to the planner buffer. Correct?

Checking with $C is something I am planning to automate in my GUI. Just not there yet.  

As for how to deal with it from the GRBL standpoint, a forced pause would also be annoying for MDI input, having to either reset or resume before being able to fix a typo in MDI. I actually have my GUI set so that I can select either the send/response or character counting protocol at will. For now I think I will just use the send/response. I have done some testing and except for programs with a ton of really small motions, the character counting doesn't really benefit. The exception to this is when I do testing on my OLD win 98 machine and then the character counting seems to help the GUI with keeping up.  

I guess my recommendation would be that if the send/response isn't affected by this, which I don't think it is, I would just leave it alone. I think most of the GUI's are just using send/response anyway. If you do decide to put a temporary hold feature in until a more permanent solution can be worked out, I would request that you make it so that a user has a compile time setting to turn this on or off.

Thanks for all the hard work you do on GRBL. I for one appreciate your efforts, and the efforts of those that have gone before!!!

chamnit, 2015-06-17:
@109JB : Correct. This problem won't affect the "send-response" simple protocol. You just won't be able to stream as fast as possible. In practice, "send-response" is fast enough, especially if you change the baud rate to 250000. 

I don't plan on installing a force pause anytime soon, and if I do, I will make it configurable to avoid an annoyed mob. I'll just make note of this in the interface wiki as a known issue. at some point.

langwadt, 2015-06-17:
At one point I tacked a number of bytes in the buffer at the end of the response so I always knew how much was safe to send

109JB, 2015-06-18:
@langwadt 
Does this have something to do with the issue being discussed as I don't see how it relates?

chamnit, 2015-07-02:
@109JB : I've got some time off for the next couple of weeks and plan on addressing a lot of the development backlog and push things forward. FWIW, I'm looking into re-factoring the communication protocol a bit to fix this problem. As well as being able to accept more real-time commands for things like overrides.

casiobearing, 2016-03-26:
I would agree with getting Grbl to buffer hold on error. The GUI I am currently developing strips out non-implemented Grbl GCode to process them locally, so such an error mechanism would work well for me.