grbl/grbl #741: Hang in 'Hold' - simplest gcode to reproduce

dirkx, 2015-06-18:
Grbl (master 0.9i, build against todays git checkout) occasinally hangs.

Attached is the simplest gcode which will reproducably hang it at around statement 3 (with up to statement 19 in the buffer). 

'''{{{

M5
M3
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X385.7143 Y10.0000 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X385.7143 Y0.0000 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X385.7143 Y10.0000 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X391.9048 Y10.0000 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X385.7143 Y5.2381 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X389.5238 Y5.2381 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X394.2857 Y6.6667 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X394.2857 Y0.0000 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X394.2857 Y3.8095 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X394.7619 Y5.2381 F50.0000 (cut)
G01 X395.7143 Y6.1905 F50.0000 (cut)
G01 X396.6667 Y6.6667 F50.0000 (cut)
G01 X398.0952 Y6.6667 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X400.0000 Y3.8095 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X405.7143 Y3.8095 F50.0000 (cut)
G01 X405.7143 Y4.7619 F50.0000 (cut)
G01 X405.2381 Y5.7143 F50.0000 (cut)
G01 X404.7619 Y6.1905 F50.0000 (cut)
G01 X403.8095 Y6.6667 F50.0000 (cut)
G01 X402.3810 Y6.6667 F50.0000 (cut)
G01 X401.4286 Y6.1905 F50.0000 (cut)
G01 X400.4762 Y5.2381 F50.0000 (cut)
G01 X400.0000 Y3.8095 F50.0000 (cut)
G01 X400.0000 Y2.8571 F50.0000 (cut)
G01 X400.4762 Y1.4286 F50.0000 (cut)
G01 X401.4286 Y0.4762 F50.0000 (cut)
G01 X402.3810 Y0.0000 F50.0000 (cut)
G01 X403.8095 Y0.0000 F50.0000 (cut)
G01 X404.7619 Y0.4762 F50.0000 (cut)
G01 X405.7143 Y1.4286 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X414.2857 Y10.0000 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X414.2857 Y0.0000 F50.0000 (cut)
G01 Z5.0000 F200.0000 (Lift to safe retraction height.)
G01 X414.2857 Y5.2381 F600.0000 (move)
G01 Z-3.0000 F10.0000 (plunge to cutting depth)
G01 X413.3333 Y6.1905 F50.0000 (cut)
G01 X412.3810 Y6.6667 F50.0000 (cut)
G01 X410.9524 Y6.6667 F50.0000 (cut)
G01 X410.0000 Y6.1905 F50.0000 (cut)
G01 X409.0476 Y5.2381 F50.0000 (cut)
G01 X408.5714 Y3.8095 F50.0000 (cut)
G01 X408.5714 Y2.8571 F50.0000 (cut)
G01 X409.0476 Y1.4286 F50.0000 (cut)
G01 X410.0000 Y0.4762 F50.0000 (cut)
G01 X410.9524 Y0.0000 F50.0000 (cut)
G01 X412.3810 Y0.0000 F50.0000 (cut)
G01 X413.3333 Y0.4762 F50.0000 (cut)
G01 X414.2857 Y1.4286 F50.0000 (cut)
M9

dirkx, 2015-06-18:
The state (reported by ?) goes from 'Run' to 'Hold' -- no limits are triggered.

chamnit, 2015-06-18:
@dirkx : Running it now and no problems here. Do you have anything connected to the feed hold pin? It could be EMI. I'd run this job without anything connected to the Arduino to see if it's your electronics or not first.

109JB, 2015-06-19:
just for kicks, I also ran the above code through an Arduino Mega and an Arduino Nano, both running GRBL 0.9i.20150618 downloaded today and both ran the code just fine with no errors.

dirkx, 2015-06-19:
Thanks. So I am running this on a barebones Arduino - with nothing connected. Will experiment with pull up resistors on the A1,2,3 pins to verify.

chamnit, 2015-06-19:
@dirkx : Can you please describe your setup? I'll need to know as much as possible to do a direct reproduction of this issue.
- Are you using an Arduino Uno or some other variant
- How are you compiling and uploading Grbl? What version of the Arduino IDE, if that is your main method?
- Please provide a print out of your Grbl settings.

dirkx, 2015-06-19:
As per below. Have since checked pullup resistors to A1/2/3. Am now checking power (dips) and similar.

* Right now cleanest stetup is an Arduino UNO (Revision r3) with nothing attached (relying on internal pull ups).

* compile from master:git - using 'make' against avr-gcc (GCC) 4.8.1, burned with avrdude version 6.0.1 with a 'make flash'. With the only setting change is setting PROGRAMMER="-P /dev/cu.usbmodemfd3231 -c arduino'.

* Grbl settings as per default:
14:21:27:  Grbl 0.9i ['$' for help]
14:21:27:  $$
14:21:27:  $0=10 (step pulse, usec)
14:21:27:  $1=25 (step idle delay, msec)
14:21:27:  $2=0 (step port invert mask:00000000)
14:21:27:  $3=0 (dir port invert mask:00000000)
14:21:27:  $4=0 (step enable invert, bool)
14:21:27:  $5=0 (limit pins invert, bool)
14:21:27:  $6=0 (probe pin invert, bool)
14:21:27:  $10=3 (status report mask:00000011)
14:21:27:  $11=0.010 (junction deviation, mm)
14:21:27:  $12=0.002 (arc tolerance, mm)
14:21:27:  $13=0 (report inches, bool)
14:21:27:  $20=0 (soft limits, bool)
14:21:27:  $21=0 (hard limits, bool)
14:21:27:  $22=0 (homing cycle, bool)
14:21:27:  $23=0 (homing dir invert mask:00000000)
14:21:27:  $24=25.000 (homing feed, mm/min)
14:21:27:  $25=500.000 (homing seek, mm/min)
14:21:27:  $26=250 (homing debounce, msec)
14:21:27:  $27=1.000 (homing pull-off, mm)
14:21:27:  $100=250.000 (x, step/mm)
14:21:27:  $101=250.000 (y, step/mm)
14:21:27:  $102=250.000 (z, step/mm)
14:21:27:  $110=500.000 (x max rate, mm/min)
14:21:27:  $111=500.000 (y max rate, mm/min)
14:21:27:  $112=500.000 (z max rate, mm/min)
14:21:27:  $120=10.000 (x accel, mm/sec^2)
14:21:27:  $121=10.000 (y accel, mm/sec^2)
14:21:27:  $122=10.000 (z accel, mm/sec^2)
14:21:27:  $130=200.000 (x max travel, mm)
14:21:27:  $131=200.000 (y max travel, mm)
14:21:27:  $132=200.000 (z max travel, mm)
14:21:27:  ok
14:21:27:  M5

Build setup

#!/bin/sh
set -e
rm -rf grbl
git clone https://github.com/grbl/grbl.git 
cd grbl
make all
PROGRAMMER="-P /dev/cu.usbmodemfd3231 -c arduino" make flash
exit 0

chamnit, 2015-06-19:
@dirkx : I don't see anything unusual, except that I don't use `make flash`. 

Here's my flashing script:
```
#/bin/bash
AVRPATH=/Applications/Arduino.app/Contents/Java/hardware/tools/avr
$AVRPATH/bin/avrdude -C$AVRPATH/etc/avrdude.conf -pm328p -carduino -P$1 -D -Uflash:w:grbl.hex
```

I'll see what happens if I do what you do.

dirkx, 2015-06-19:
Thanks. Hold your horses though; I am experimenting with swapping boards & Atmega chips - and have now one chip which is different regardless of board. So it may be hardware related.

dirkx, 2015-06-21:
Se we tried with different boards (all revision r3); and swapped different batches of Atmega 328P rev. D (DIL sockets) around:

- None of them exhibits this issue with the GrblController.

- None has the issue at 9600 baud.

- We've seen a few cases where tehe Atmega*u2 (the usb to serial) is at fault - i.e. the snoop on the TTL shows missed transitions.

- Some batches of AtMEGA (1437 marking) will *not* give the issue when fed commands with 0.2 Second delay; will give the 'Hold' hang when fed normally; and will give sporadic ' error: Invalid gcode ID:24' and 'error: Bad number format' when fed at 115200 'max speed' - i.e. the next command within 12 milli Second after the \n\r of the ok.

- The ones with 0723 will randomly give a Hold regardless of baudrate / spacing.

- The ones marked 1004 will give the Bad number/invalid opcode only when fed faster than with a 100mSecond spacer; never the Hold.

So our guess right now is that at 115200 we are hitting a limit; first on some of the 328P's and in a few cases something in the USB/TTL; but much less so.

chamnit, 2015-06-21:
In the past, Grbl controller has had problems with its streaming protocol. I would suggest trying something like UGS, bCNC, or the Grbl streaming Python script. The latter is the simplest and removes GUI influences on troubleshooting problems like these.

chamnit, 2015-07-02:
@dirkx : Is this issue resolved? 250000 baud can be used as well and should have less timing error than 115200, but in practice, 115200 hasn't ever been an issue.

dirkx, 2015-07-06:
@chamnit: correct - we are not using GrblController (it is far too slow and error prone) - but instead stream to the serial socket directly (https://github.com/dirkx/HersheyEngravingFontsForCocoaAndiPhone).

@chamnit: We cannot reproduce the issue at 250000 with any of the chips.

dirkx, 2015-07-06:
Closing the issue - as it is serial timing related.

closed 2015-07-06 by dirkx