grbl/grbl #742: 0.9i - Min feedrate appears to be around 40-45mm/min

jarretluft, 2015-06-19:
When a low F-value is specified (under 45mm/min or so), GRBL will move a minimum speed of around 43mm/min.
This was tested by setting a feedrate of F5 mm/min, and then attempting a move of G1 X5 from the work origin.  The machine moved 5mm in about 7 seconds but this should have taken 1 minute at 5mm/min.

More detailed report here: http://www.shapeoko.com/forum/viewtopic.php?f=3&t=6103

For testing this I am running: $i = [0.9i.20150329:]

109JB, 2015-06-19:
Was curious about this since I did feedrate testing when I started down the GRBL road, so I ran it on 0.9i.20150618 and could not duplicate the problem. Still curious, I then downloaded the 0.9i.20150329 version and still couldn't duplicate it. In both cases, I set F5 and then G90G1X5 starting from x=0 and it took exactly 1 minute to complete the move. did the same for the Y and Z axes too.

sacidu93, 2015-06-19:
This behaviour appears only when enabling the AMASS  feature in config.h

#define ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING // Default enabled. Comment to disable.

as the AMASS algorythm requires only one timer prescalar; very low feedrate will not be guaranteed with only 0xffff value loaded in Timer ( prep_segment->cycles_per_tick=0xffff // line 786 stepper.c) 
AMASS level 3 is equivalent to a prescallar of 1:8 which is not the case when disabling AMASS and using the same value 0xffff with 1:64 prescalar.

chamnit, 2015-06-19:
It's been a while since I did this, but I recall that I took into account for very slow step segments being generated and executed. It shouldn't be doing this. That said, I may have not looked at all of the settings. @jarretluft : Please provide the settings you have to reproduce this issue.

jarretluft, 2015-06-19:
On shapeoko 2 hardware, Arduino uno w/ grblshield v4.
Using the stock grbl [0.9i.20150329:] with no customization.

$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=2 (dir port invert mask:00000010)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=3.000 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=350.000 (homing seek, mm/min)
$26=254 (homing debounce, msec)
$27=2.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=1280.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=250.000 (z max rate, mm/min)
$120=500.000 (x accel, mm/sec^2)
$121=500.000 (y accel, mm/sec^2)
$122=30.000 (z accel, mm/sec^2)
$130=295.000 (x max travel, mm)
$131=306.000 (y max travel, mm)
$132=96.000 (z max travel, mm)

chamnit, 2015-06-19:
@jarretluft : Thanks Jarret. I've verified that this is related to a low step/mm setting. To fix this correctly will take some significant refactoring. Since, moving this slow on a mill isn't common, I'm not going to touch this for the v0.9 master, but slate it to be fix in v1.0.

chamnit, 2015-07-24:
Ok, I've located the cause of this issue. It stems from the AMASS algorithm and it only using a fixed prescaler in the main timer. AMASS' lowest tick rate is 26 step/sec. This translates to roughly 40mm/min for 40step/mm. There isn't much I can do to fix this with AMASS due to way it was designed. I was not expecting use cases that requires really slow feed rates. 

So the solution is to disable AMASS in config.h, which forces Grbl to use the old timing algorithm, and/or increase the steps/mm of the axes, such that the lowest desired step rate is above 26 steps/sec.

closed 2015-07-24 by chamnit