grbl/grbl #744: Z axis stall immediately after XY Move

bensuffolk, 2015-06-20:
This is a strange issue, and I suspect it might not be grbl itself, but let me explain it and see what you think.

I am running 0.9i nad using grblweb for control.

If I issue the following commands (from a gcode file generated by jscut) :

G21         ; Set units to mm
G90         ; Absolute positioning
G1 Z10 F2000      ; Move to clearance level
G1 X101.8197 Y26.3550 F2000
G1 Z0.0000

The G1 Z0 command always stalls as it starts, but about half way through starts to work. Leaving the Z Axis about 2-3mm higher than it should be. However I can bounce Z axis up and down repeatedly with no problem by doing the G1 Z10 followed by G1 Z0. If I make a file with those commands repeated 10 times its all fine.

I can Move X Y and Z Axis all at the same time with no problems. Its just Z axis immediately after and X Y Move. 

I'm away today with work, but when I get home I am going to try :

1. swapping the Z stepper driver with a spare I have in case its related to that.

2. Moving Just X then Z

3. Moving Just Y then Z

4. A slower feed rate for Z, but since it works fine not directly after a XY move I don't think its this.

Other things that may be relevant. Its a Twin Motor System on X, Single on Y and Z. Each is 1.2A and the drivers are set for that (DRV8825, with current control voltage set at 0.6v). The PSU is 24V @7.5A 

Because I cam move all 3 axis at the same time, it does not seem like it should be a power issue, but could anything else cause this sort of problem?

chamnit, 2015-06-20:
It's likely a settings issue with the z-axis. When you move all three axes, Z isn't moving as fast or accelerating as much as it would on its own. Reduce the Z-acceleration and max velocities by half to see if that fixes your problem. Also, check you junction deviation setting. It should be around 0.01-0.02 mm.

bensuffolk, 2015-06-20:
I'll try those settings and see if that works. Would the acceleration affect the fact it's only an issue after a move and not a problem when run from stationary?

Ben

chamnit, 2015-06-20:
It's likely due to the junction deviation setting. The higher it is, the faster it'll go through consecutive motions. See wiki for detuials on how it works.

bensuffolk, 2015-06-20:
I checked the junctino deviation and for some reason it was 2000! I set it to 0.01 and all is good without changing any of the acceleration or max speed settings.

I'm not sure how it ended up being that value, but clearly that was the problem.

Many thanks

closed 2015-06-20 by bensuffolk