grbl/grbl #746: AT90USB support and USB serial

imrahil, 2015-06-23:
Hi all!
thanks to your hard work and awesome external cpu_maps couple of days ago I added support for boards with AT90USB to latest grbl (I based on fork from lincomatic from last year https://github.com/lincomatic/grbl/). It should works with Printrboard but my goal was KOS board (http://www.byq3d.eu/index.php?id_product=18&controller=product&id_lang=7)

I had to disable all controll pins because at90 has only one pin change interrupt register and it's blocked by disable and limit pins...

Except that almost everything works fine but I have two problems:
- stable connection from other gcode senders than Universal Gcode sender (https://github.com/winder/Universal-G-Code-Sender/) or default python one (simple_stream.py)... For example I cannot connect from grblPanel or Chillipeppr (I receive only "Expected command letter" errors)
Probably I messed too much with serial connection but I'm not C/C++ ninja so I'm stuck at the moment...

- disabling motors on different ports/pins. at the moment my Y motor is always on.

Cheers!

ps. I love arcs/circles with G2 on my CoreXY!! :) (I'm newbie in CNC, had some 3d printing experience)

chamnit, 2015-06-23:
@imrahil : Thanks for sharing! I've always wanted to port Grbl to the AT90USB at some point so that I can get it running on my Printrboard, just as you said. Keep us posted on your progress!

imrahil, 2015-06-29:
Hi! I think I figured out how this damn USB works :) moved some initialization code into main loop because AT90USB doesn't restart when serial connection starts.

Special chars (status report, cycle start, etc) aren't catched on interrupt right now but I think that USB in at90 is so fast that it's no difference

Universal-gcode-sender works fine now

valpoder, 2015-11-09:
Hello imrahil:

Im also very interested on running GRBL on the AT90USB for Printrboard.

Would it be possible sharing your code with us ???
did you compile using arduino IDE or a makefile (avr-gcc) ??

Thanks in advance !!!

imrahil, 2015-11-09:
Hi! Funny thing! I didn't mention about my fork nowhere! :) Code is available here - https://github.com/imrahil/grbl/tree/at90usb1286_support
Compiled like normal grbl but with Teensyduino support (https://www.pjrc.com/teensy/td_download.html)

I'm pretty sure that Printrboard has different pin layout than my board so be carefull!

valpoder, 2015-11-09:
thats very good news !!! thanks for the link ....
>> I'm pretty sure that Printrboard has different pin layout than my board so be carefull!
dont worry , ii will work on the pinout... maybe we can add it to your git later on if it work..  !!
cheers !!!

valpoder, 2015-12-18:
imrahil:   i added cpu map for printerboard ...
need to add option to makefile & add a cpu_map file....

would you want to add this to your fork ?

imrahil, 2015-12-19:
> it looks like your branch is broken...

how it's broken?

valpoder, 2015-12-21:
if i make a clean chekout and then make i get:::

$ make
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/main.c -o build/main.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/motion_control.c -o build/motion_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/gcode.c -o build/gcode.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/spindle_control.c -o build/spindle_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/coolant_control.c -o build/coolant_control.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/serial.c -o build/serial.o
grbl/serial.c:137: warning: 'USART0_UDRE_vect' appears to be a misspelled signal handler
grbl/serial.c:191: warning: 'USART0_RX_vect' appears to be a misspelled signal handler
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/protocol.c -o build/protocol.o
avr-gcc -Wall -Os -DF_CPU=16000000 -mmcu=atmega328p -I. -ffunction-sections -fdata-sections -MMD -MP -c grbl/stepper.c -o build/stepper.o
grbl/stepper.c: In function 'st_wake_up':
grbl/stepper.c:199: error: 'PORTE' undeclared (first use in this function)
grbl/stepper.c:199: error: (Each undeclared identifier is reported only once
grbl/stepper.c:199: error: for each function it appears in.)
grbl/stepper.c: In function 'st_go_idle':
grbl/stepper.c:256: error: 'PORTE' undeclared (first use in this function)
grbl/stepper.c: In function '__vector_11':
grbl/stepper.c:328: error: 'PORTA' undeclared (first use in this function)
grbl/stepper.c: In function '__vector_16':
grbl/stepper.c:459: error: 'PORTA' undeclared (first use in this function)
grbl/stepper.c: In function 'st_reset':
grbl/stepper.c:507: error: 'PORTA' undeclared (first use in this function)
grbl/stepper.c: In function 'stepper_init':
grbl/stepper.c:516: error: 'DDRA' undeclared (first use in this function)
grbl/stepper.c:518: error: 'DDRE' undeclared (first use in this function)
Makefile:53: recipe for target 'build/stepper.o' failed
make: *** [build/stepper.o] Error 1

this might be caused by wrong cpu in makefile

valpoder, 2015-12-21:
i think i get it now...  
 the updated mak efile is missing usb file.. do this and it will work...

DEVICE     ?= at90usb1286
CLOCK      = 16000000
PROGRAMMER ?= -c avrisp2 -P usb
SOURCE    =  usb_serial.c main.c motion_control.c gcode.c spindle_control.c coolant_control.c serial.c \
             protocol.c stepper.c eeprom.c settings.c planner.c nuts_bolts.c limits.c \
             print.c probe.c report.c system.c

I made a version for the printrboard

imrahil, 2015-12-21:
sorry, I've never used a makefile :) always compiled via Arduino IDE

theshortest, 2016-12-09:
Any chance someone could share a pre compiled Hex for Printrboard RevD. Tried Lincomatic's and I cannot get it to connect.