grbl/grbl #747: GRBL freeze

MichielD99, 2015-06-24:
When grbl is engraving a drawing with my laser engraver, at a random moment, it freezes. My engraver doesn't move anymore and the laser stops. Is there something I can do?
Thanks.

chamnit, 2015-06-24:
You're going to have to be a bit more elaborate of your problem. What Grbl version are you using. Describe your setup. Where in the program is it having a problem. Is it reproducible. Does the program run without anything connected to the Arduino to eliminate electrical influences. Etc, etc, etc.

MichielD99, 2015-06-24:
I am using grbl v0.9i and those are my settings:

$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=50.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=244 (homing debounce, msec)
$27=3.000 (homing pull-off, mm)
$100=133.333 (x, step/mm)
$101=133.333 (y, step/mm)
$102=250.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=500.000 (x max travel, mm)
$131=400.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

I have nothing connected to the arduino.
But when i sent it the exactly same file, it runs without any errors. I have this problem a lot.

chamnit, 2015-06-24:
@MichielD99 : Then it's most likely not a problem with Grbl. It's an issue with your electronic wiring or some other external electrical interference. Use a process of elimination to figure out what component is causing the issue. Plug in one thing in at a time and re-run to try to re-create the problem. Outside this, there isn't much I can do to help you.

closed 2015-06-24 by chamnit

MichielD99, 2015-06-24:
Ok! I will do that.
Thank you for your time and your advice!