grbl/grbl #748: Adding SPI

J-Dunn, 2015-06-24:
I'm trying to add some hardware using driver chips programmed via SPI.  Mostly initialisation but also error status.  Usual step and dir lines. 

Arduino SPI lib is pretty trivial since most of it is done by the hardware. It's mostly poke a byte into a register and wait for a status flag, then read the return byte. 

I'll need to shift the default pins around a bit to free up MOSI and friends, so I won't have spindle control ( not needed ) and limit pins will have to go onto port A. 

Now I know GRBL has been squeezed to a minimum to get it to fit into a UNO. How much room do I have left to play with before it starts giving problems? What are the limits I need to pay particular attention to? 

Thanks  JD.

chamnit, 2015-06-24:
@J-Dunn : You want to have about 400 bytes of memory free and don't want to exceed 31-31.5kB of total flash (boot loader resides in the last 0.5kB of 32kB). Or you can just use the Mega2560 cpu map to get extra flash, memory, and pins.

J-Dunn, 2015-06-24:
"Binary sketch size: 27,962 bytes (of a 32,256 byte maximum)"

Many thanks. So I still have some breathing space but not much. I have a nagging doubt that sketch size does not account for stings stored in pgmspace. Can you comment?

chamnit, 2015-06-24:
It accounts for everything. Pgmspace is flash stored.

closed 2015-06-24 by chamnit

dgoadby, 2015-06-25:
I'm just about to build my first GRBL unit and noted the comment in this thread about using the Mega2560. I was just about to order one when I looked at the physical I/O layout. I was going to use the CNC shield with plug-in stepper drivers but The Mega2560 is way different. I guess the number of I/O lines involved isn't high but is it worth the effort?

chamnit, 2015-06-25:
@dgoadby : Grbl on Mega2560 is not officially supported, the pin mapping isn't compatible with existing shield (i think) and there may be bugs. So tread carefully.

langwadt, 2015-06-26:
I'm guessing something like the L6474?
I'm not sure how it would affect code size but you could just bit bang the spi, it is much more flexible with pinout and the speed  shouldn't be a big issue since you would generally only use it for setup and/or when there is an alarm and you have to stop everything anyway

J-Dunn, 2015-06-26:
yes, L6474. Thanks for the suggestion, but ...

Arduino hardware handles all the SPI interaction. Just write a register and read back the return byte. Doing it by hand woud be considerable larger and basically be a waste of development time. It would allow using different pins but you still need four of them.  I'll just have to move the limit switches and drop some of the extra functionality like spindle speed. 

This is a stepping stone to moving the whole thing to stm32 anyway.  I just did not fancy throwing all the balls into the air at once. Using unknown driver chips, a new mirco and porting GRBL all in one hit seemed like a recipe for major debugging headaches.

J-Dunn, 2015-06-26:
I've stripped the necessary bits out of Michel_F's L6474 library and SPI lib , dumping all cpp classes. 
With a token call to each fn to ensure linking , it surprisingly light. 
Binary sketch size: 27,962 bytes (of a 32,256 byte maximum)
Binary sketch size: 28,726 bytes (of a 32,256 byte maximum)

This is not integrated with GRBL yet but as footprint feasibility it looks promising. 

Now I have to resolve the pinouts. Since the ST boards were intended to use PWM for simplistic motor driving.  The main worry is, while all the dir pins are on portD, one of  the step pins is on portB. so I'll either have to bend some pins or bend my GRBL.

I'll try the latter first.

```  // Define step pulse output pins. NOTE: All step bit pins must be on the same port.```
This seems to because it allows some nice optimisation in the critical ISRs. If I'm on split ports it's going to basically double the code. 

It looks like the major hit is going to be in ISR(TIMER0_OVF_vect)

ISR(TIMER1_COMPA_vect) is already  a bit more loaded, so perhaps less of a %age hit. 

Is there anything here that is going to be a real killer and make be want to bend the hardware?

thx.

chamnit, 2015-06-26:
@J-Dunn : No. Most these 8-bit masks and register settings take only a couple of cycles per operation. You'll probably won't notice any difference if you stick to simple 8-bit manipulation only. The main reasons that I haven't separated them out are that there wasn't a need to and it save a bit flash and a few cycles.

J-Dunn, 2015-06-26:
Trying modify stuff I find that there does not seems to be a centralised structure to the include chain. 
For example default.h has the following conditional but does not include config.h where it is defined. 
``` #ifdef DEFAULTS_GENERIC ```
I also expected to defining N_AXIS=2 to do something in relation to some  #ifdefs in the code. 
I was hoping to build and two-legged GRBL to get around the pin issue to test what I have so far.

That kind of thing would nice, and fairly simple.

chamnit, 2015-06-26:
@J-Dunn : Yep. That king of thing would be nice.

J-Dunn, 2015-06-26:
maybe at least defaults.h should #include config.h

J-Dunn, 2015-06-27:
Due to pin shuffliing I need to have limit pins on portC as well as the external user i/p pins. 

This means some trickery since they then have the same ISR.  I have made a pseudo ISR that is called from the true ISR and does exactly the same thing. 

```
// if limit sw and Ext input pins are on same port , handle ext pins subroutine
#if  ( LIMIT_INT_vect == EXT_INPUT_INT_vect )

uint8_t  Ext_pin_peudo_ISR() 
{
  // Enter only if any  pin is actively low.
  if ((EXT_INPUT_PIN & EXT_INPUT_MASK) ^ EXT_INPUT_MASK) { 
    if (bit_isfalse(EXT_INPUT_PIN,bit(PIN_RESET))) {
      mc_reset();
    } else if (bit_isfalse(EXT_INPUT_PIN,bit(PIN_FEED_HOLD))) {
      bit_true(sys.execute, EXEC_FEED_HOLD); 
    } else if (bit_isfalse(EXT_INPUT_PIN,bit(PIN_CYCLE_START))) {
      bit_true(sys.execute, EXEC_CYCLE_START);
    } 
  }
  return 1;
}
#else 
uint8_t Ext_pin_peudo_ISR() { return 0; }
#endif
```
```
#ifndef ENABLE_SOFTWARE_DEBOUNCE
  ISR(LIMIT_INT_vect) // DEFAULT: Limit pin change interrupt process. 
  {
    // Ignore limit switches if already in an alarm state or in-process of executing an alarm.
    // When in the alarm state, Grbl should have been reset or will force a reset, so any pending 
    // moves in the planner and serial buffers are all cleared and newly sent blocks will be 
    // locked out until a homing cycle or a kill lock command. Allows the user to disable the hard
    // limit setting if their limits are constantly triggering after a reset and move their axes.
    if (sys.state != STATE_ALARM) { 
      if (bit_isfalse(sys.execute,EXEC_ALARM)) {
        mc_reset(); // Initiate system kill.
        bit_true_atomic(sys.execute, (EXEC_ALARM | EXEC_CRIT_EVENT)); // Indicate hard limit critical event
      }
    }
    if ( Ext_pin_peudo_ISR() ) { return;}
 }  
```
The  true ISR in system.c gets removed with a similar #if 
It means  pulling a few lines of code from system.c in to limits.c  but that is not a big deal. 

Does the logic look sound? 

Thx.

J-Dunn, 2015-06-27:
It seems that there is dead variable in the settings struct. I was wondering what it could be for and found out : nothing ;)

settings.dir_invert_mask

It gets printed by report.c , it can be set from command line, but it's never used. Step masks need inverting to end the step pulse, but directions are inverted one by one. according to position. 

A little clean up that can be done when the occasion arises .

chamnit, 2015-06-27:
You sure about that? Check stepper.c's initialization routines. I think you'll find that it's not true.

J-Dunn, 2015-06-27:
Well it does get stored in EEPROM but it seems to have no use. I've just edited out all mention of it and removed it from the struct and it compiles, so I must have reviewed all usage.  It saved 132 bytes. 

It seems to have been cloned the step mask in  st_generate_step_dir_invert_masks() but never got used.

chamnit, 2015-06-27:
@J-Dunn : I don't see any such cloning of the direction invert into the step mask. They are completely separate entities. Both of which are used properly for their particular purpose. Could you point out where you think dir_invert_mask is not being used in the source code?

J-Dunn, 2015-06-27:
OK, I think I've seem what I was missing. Each time ISR(TIMER1_COMPA_vect) fires it inverts which steppers get treated: either the principal one or the subordinate Bresenham axes.

I got mislead because there did not seem to be any way that these direction masks could change programatically.  It seems that the whole of point these not being defined constants is to allow it to be set interactively and stored. It takes while to realise what some of this is doing and it's confusing because several of these variable names are similar. 

Hopefully I'm with it now. Sorry for the noise.

J-Dunn, 2015-06-28:
I'm getting a better understanding of inner workings of the ISRs now.You have stated that one of your aims is to make this code easy to adapt and I would agree it is well commented and well laid out. One thing I have found a barrier is the variable names, which are often confusing. I will submit some suggestions once I'm more familiar with the code. For the moment I'm making changes locally and taking notes.

One thing struck me as a bit risky. 
```   STEP_PORT = (STEP_PORT & ~STEP_MASK) | st.step_port_bits; ```

This will affect the bits on DIRECTION_PORT which have only just been set "a couple of ns" earlier, since both step and dir are on the same Arduino port by default. The values output here are determined at the end of the previous call to this ISR. :
   st.step_port_bits ^= step_port_bit_mask; 

To be accurate this line should only XOR the relevant bits or the output line should OR with only the relevant bits. eg.
```    STEP_PORT = (STEP_PORT & ~STEP_MASK) | (st.step_port_bits & STEP_MASK );```

It costs 2bytes. 

Recognising that GRBL is mature and well tested software, it seems to work in the field, though it seems to be flying close to the wire to  accidentally  change the dir port at the same time as setting STEP_PORT output. It also does not aid understanding the code and portability

Contrariwise, the ISR which resets the pins maybe doing more than needed. 
```  STEP_PORT = (STEP_PORT & ~STEP_MASK) | (step_port_bit_mask & STEP_MASK); ```

 step_port_bit_mask may contain more or less bits than the defined constant STEP_MASK if set by manual command line or EEPROM. If more, STEP_MASK will kill extra bits (which is fine) but this should not be caught here. If less, STEP_MASK is redundant.

This highlights that manual input could activate an axis for which no pins are defined. this should be trapped at source, not in the ISR.

J-Dunn, 2015-06-28:
```  // regroup all these ifdefs. 
#ifdef ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING
    st.counter_x += st.steps[X_AXIS];
    st.counter_y += st.steps[Y_AXIS];
    st.counter_z += st.steps[Z_AXIS];
#else
    st.counter_x += st.exec_block->steps[X_AXIS];
    st.counter_y += st.exec_block->steps[Y_AXIS];
    st.counter_z += st.exec_block->steps[Z_AXIS];
#endif 
```

chamnit, 2015-06-28:
You're getting into semantics, which are welcome, but this can easily become a time drain without much benefit. Good catch on the step mask part. It's implied to be correct but isn't portable as you say. It probably was an oversight when it was written. As for combining the counters, it doesn't matter.

J-Dunn, 2015-06-28:
counter was just a trivial tidy up that makes it easier to read. Glad I was correct about STEP_MASK thing, it means I'm getting to grips with what it's doing. 

 #define DEFAULT_DIRECTION_INVERT_MASK ((1<<Y_AXIS)|(1<<Z_AXIS))
The default value here is 6 and it is NOT an inversion mask. Confusing. May be helpful to call  it a map to distinguish it from the masks, it's functionally different. I will suggest better names when I get through. 

I'm having a little trouble getting why the direction of the subordinate axes is opposite to the driving axis. Hint?

J-Dunn, 2015-07-01:
@langwadt  You seem familiar with the L6474 and  you said you prefer to  bit-bang the SPI. Have you had a look at the L6474 lib posted by Michel F ?
https://github.com/MotorDriver/L6474

It looks fairly thorough but I'm not getting sensible status bytes back from SPI using the demos he provides. The demos make the motors move but I'm not sure that SPI is working correctly.

I was hoping to use that as a base for bringing L6474 to GRBL and kinda assumed it was working code.

langwadt, 2015-07-01:
yes I use an L6474 in one project at work and I also have a port of grbl running on Nucleus stm32f411 + 3*L6474.  
I've seen the lib and it has the strength and weakness such libs always have, they are so 
generic. I'm doing it much more simple, one function writes the bytes needed to setup all 
the registers, one function to enable/disable outputs, one function to read status  

as for the status, remember that many of the status bits are active low

J-Dunn, 2015-07-02:
thanks langwadt, would you be willing to share the Nucleo port of grbl? That is where I'm heading. Getting this working on Arduino was just a first step to master daisy-chaining three boards using example code in the form of that library. It's obviously far too clunky to use as a lib in this context. I've  stripped the guts out it to add to grbl and the initial build only added about 800 bytes. 

I found that I was not getting correct returns from SPI so went back to the library demos and found that the same thing was happening there. So I'm not sure whether the lib is buggy or I have a wiring fault. 

If you already have this working on stm32f4  it would save me a lot of time reinventing that particular wheel. Thanlks.

J-Dunn, 2015-07-05:
I'm close to having GRBL + L6474 working on top of Arduino. 

Obviously there is some extra overhead and GRBL is pretty tight already. Do you have any stress tests that I can run?  I'm especially thinking of stack and heap overflow when I connect the hardware. 

This is quite a heavy machine and I don't want something like this showing up during hardware testing. 

I could invent some tests but with the maturity of the probject I imagine that someone has already done this. Any suggestions? 

Thanks.

J-Dunn, 2015-07-23:
I have got a stack of three L6474 working with GRBL, code size 30440 bytes. Could be trimmed a little but that's well within the code limit.

It does a hard stop , like a limit pin, if there's a critical error that puts the outputs in HiZ state;  a soft stop for a thermal warning; and just clears non critical errors like bad command bytes. I've used INT0 since that is how the L6474 are wired. By turning sei() back on at the top of the ISR it causes minimal disruption to GRBL timing. 

All that is subject to proper testing but the simple circle.gcode example confirms GRBL is still functioning.

Mashfique, 2017-03-09:
Hi @J-Dunn : I'm also trying to run GRBL with powerSTEP01 which is  similar to L6474 but with a higher current rating. Can you please share your code or project that worked for Arduino+GRBL+L6474. I'm also planning to go for STM32s in future. but just wanted to try Arduino first as I have used Arduino+GRBL_Easy Driver to run a small CNC.

Thank you and the rest of the people here. your discussion have been very enlightening. 

-Kazi

J-Dunn, 2017-03-09:
Hi, I don't have anything readily in a publishable state, it's all very much WIP.  I'll give it some thought if I have a little time in the near future. 

I'm currently porting the stm32 code to work with the STEP01 boards too,  though I have been trying to retain the previous code base by suitable compiler swiches it would likely a little debugging before it worked on AVR again. 

The step01 code is ready for testing on the hardware and I can flip it from L6474 to Step01 by changing one #define.

Mashfique, 2017-03-10:
Alright. Thanks @J-Dunn. 
Will be checking your posts regularly .

J-Dunn, 2017-03-18:
So as not to leave you nothing, here is what I think is the most recent version of the L6474 on AVR files. Add them as a library in the usual way.  I use arduino makefile to build and debug using Geany editor, to get out of the horrible Arduino pseudo-IDE but I think it should still work in the IDE.

It provides basic SPI communication for up to three L6474 boards on top of Arduino. The round trip for a 3 board get_status() is about 19us , which is plenty fast enough to fit inside the usual top speed for Arduino GRBL.  The code just  fits into memory with about 140b spare IIRC. 

https://github.com/J-Dunn/grbl_stm32/blob/master/grbl_L6474_latest.bz2

You will need to change the basic GRBL code to call the lib funcs are needed. Sorry I don't have a complete click and go solution to hand but there you have the guts of a fast working interface. The rest should be fairly obvious: setup the FLAG pin like a GRBL limit input , etc. 

At the end of the c file you will find an interrupt handler for the FLAG pin that sorts out the critical and non critical errors and provides some USB output. 

When I have finished debugging the STEP01 code I may have something more complete.

Mashfique, 2017-03-22:
Thanks :)

Sincerely,
Kazi Mashfique Hossain
930 Manistique St
Detroit , MI 48215
Contact No. +1-402-617-0745
Email : kazi.mashfique@gmail.com

------------------------------------------------------------
-----------------
Master of Science in Mechanical Engr. and Engr. Mechanics
Mechanical and Material Engineering
University of Nebraska - Lincoln
Contact No. +1-402-617-0745
Email : kazi.mashfique@gmail.com
           kazi.mashfique@huskers.unl.edu