grbl/grbl #750: 3 Axis Work together

Lexander24, 2015-06-26:
Hello, i want to ask.
Can i make 3 axis move together?
I try use 
G91
G01 X5 Y3 Z2 F200

But, my stepper connect with z axis didn't move.

Thank you

Norwegian-Gadgetman, 2015-06-26:
There's no reason why it shouldn't work. 3axis movements are often needed when doing tabs (holds the part in place in the material, at the bottom of cuts)
I assume that if you use just 'G1 Z2' the Z-axis moves correctly?

twforeman, 2015-06-26:
That should totally work. Can you move the Z axis by itself?

Lexander24, 2015-06-26:
Yes.
When i just use
G91
G01 Z2, the Z axis move correctly.

But i can't make combination like
XZ or YZ or XYZ

Lexander24, 2015-06-26:
For detail i use GrBl firmware version 0.9i
Arduino uno R3 clone smd

Code i try
G91
G01 Z2 F200
(The z axis move correctly)

G91
G01 X20 Y10 
(The x and y axis move correctly)

But when i try
G91
G01 X20 Y10 Z2
(Just x and y axis move correctly, but the z axis didn't move)

G91
G01 X20 Z2
(Just x axis move and z axis didn't move)

G91
G01 Y10 Z2
(Just y axis move and z axis didn't move)

Any parameter i must check? Or how to write the g code because i'm new user. Thank You

chamnit, 2015-06-26:
@Lexander24 : Everything works as intended. Please read and understand how g-code works. Documentation is all over the web.

closed 2015-06-26 by chamnit

chamnit, 2015-06-26:
@Lexander24 : And also check your settings and wiring.

109JB, 2015-06-26:
This is a strange problem, but like @chamnit said, I don't think it is a GRBL issue or everyone would be having the same problem  However,. I have been thinking about this and have some things  for you to check. The problem could be that the Z-axis is not getting enough power when the other 2 axes are running, which is causing it to essentially stall. Some questions to try to sort this out for you are:

1. How is the wiring set up for the stepper motor power? Are there individual wires from the stepper motor power supply to each driver, or is there one wire from the power supply to x, and then from there to Y, and then to Z? Each driver should have its own wire directly to the power supply.

2. What are the specs for your motors, drivers and power supply. If the power supply is too small, it can give problems.

3. What current setting do you have your drivers set for?

4. Do all of the axes move smoothly without binding

5. What type of machine is it (Milling machine, Router, PCB mill, etc.)

6. when you try to do a G91 G1 X5 Y3 Z2 F200 move, does the Z axis motor make any noise at all?

Lexander24, 2015-06-26:
Answer
1. I will change the wiring cable
2. If the problem is power. The x and y use bigger power and work ok togather. So i think when x and y can work togather. X and Z or Y and Z no problem when work together 
3. About 3.5 A for motor X and Y, and 2A for Z motor
4. All axes move very smooth and accelaration work perfectly
5. I confused when you quest about this. Because i use this concept for make robot for take something on storage
6. When i try
G91
G1 X5 Y3 Z2 F00
The z motor not make any noise or vibration

ANKUR-JANGIR, 2022-01-03:
Same problem with me