grbl/grbl #751: Homing fail after 0.9i upgrade

hamzaozturk, 2015-06-27:
I use no limit switches. Everything was fine until i upgraded to the latest release. Limit switches connected to the X-, Y+, and Z+ pins.

You can find the output of the terminal below.

[verbose]<Idle,Lim:000>

$$ Response:
2015/06/27 17:01:59 - $0=10 (step pulse, usec)
2015/06/27 17:01:59 - $0=10 (step pulse, usec)
2015/06/27 17:01:59 - $1=25 (step idle delay, msec)
2015/06/27 17:01:59 - $2=0 (step port invert mask:00000000)
2015/06/27 17:01:59 - $3=0 (dir port invert mask:00000000)
2015/06/27 17:01:59 - $4=0 (step enable invert, bool)
2015/06/27 17:01:59 - $5=0 (limit pins invert, bool)
2015/06/27 17:01:59 - $6=0 (probe pin invert, bool)
2015/06/27 17:01:59 - $10=16 (status report mask:00010000)
2015/06/27 17:01:59 - $11=0.020 (junction deviation, mm)
2015/06/27 17:01:59 - $12=0.002 (arc tolerance, mm)
2015/06/27 17:01:59 - $13=0 (report inches, bool)
2015/06/27 17:01:59 - $20=0 (soft limits, bool)
2015/06/27 17:01:59 - $21=0 (hard limits, bool)
2015/06/27 17:01:59 - $22=1 (homing cycle, bool)
2015/06/27 17:01:59 - $23=5 (homing dir invert mask:00000101)
2015/06/27 17:01:59 - $24=100.000 (homing feed, mm/min)
2015/06/27 17:01:59 - $25=500.000 (homing seek, mm/min)
2015/06/27 17:01:59 - $26=250 (homing debounce, msec)
2015/06/27 17:01:59 - $27=0.000 (homing pull-off, mm)
2015/06/27 17:01:59 - $100=1600.000 (x, step/mm)
2015/06/27 17:01:59 - $101=100.000 (y, step/mm)
2015/06/27 17:01:59 - $102=650.000 (z, step/mm)
2015/06/27 17:01:59 - $110=300.000 (x max rate, mm/min)
2015/06/27 17:01:59 - $111=300.000 (y max rate, mm/min)
2015/06/27 17:01:59 - $112=300.000 (z max rate, mm/min)
2015/06/27 17:01:59 - $120=50.000 (x accel, mm/sec^2)
2015/06/27 17:01:59 - $121=200.000 (y accel, mm/sec^2)
2015/06/27 17:01:59 - $122=50.000 (z accel, mm/sec^2)
2015/06/27 17:01:59 - $130=200.000 (x max travel, mm)
2015/06/27 17:01:59 - $131=200.000 (y max travel, mm)
2015/06/27 17:01:59 - $132=50.000 (z max travel, mm)

Thanks for your response in advance.

Scott216, 2015-06-27:
Is the Z limit switch connected to D11 or D12.  With the latest release of GRBL, the default pin for Z limit switch changed from D11 to D12.

chamnit, 2015-06-27:
@hamzaozturk : And set your $27 pull off setting to greater than zero.

closed 2015-06-27 by chamnit

hamzaozturk, 2015-06-27:
Yes it is connected to D12. Is it necessary to change $27 as greater than zero?

hamzaozturk, 2015-06-27:
First, my z axis goes up, and hits to the limit switch. Then it gives ALARM: Homing fail message.