grbl/grbl #752: Homing fail after 0.9i upgrade

hamzaozturk, 2015-06-27:
As i mentioned in the issue #751, i still get ALARM: homing fail error.

I also changed $27 = 1.

My z axis goes up, and hits to the limit switch. Then it gives ALARM: Homing fail message.

bensuffolk, 2015-06-27:
Are you sure the z axis limit switch is wired to the correct pin? I had that problem when I had the limit switches wired to the wrong pins. So the z switch was actually swapped with the x in my case. Swapped that round and all good

Regards

Ben

Ben Suffolk
ben@vanilla.net

All prices quoted exclude VAT at the current rate unless otherwise stated. E&OE.

This e-mail including any attachments is confidential and may be legally privileged. If you have received it in error please advise the sender immediately by return email and then delete it from your system. The unauthorised use, distribution, copying or alteration of this email is strictly forbidden. If you need assistance please contact us on +44 117 9117 914

hamzaozturk, 2015-06-27:
When i push to the z-limit i get the output as below. is not it right?

Lim:100

chamnit, 2015-06-27:
@hamzaozturk : Please stop opening new issues threads. Make sure your pull-off setting is greater than one. And from the looks of it your homing invert for the z-axis is wrong if you have it set as Z+. There should be no invert. Please check all of your settings and check your limit switch wiring by measuring the voltage.

chamnit, 2015-06-27:
@hamzaozturk : That means your z-limit switch is NOT normally-open.

closed 2015-06-27 by chamnit

109JB, 2015-06-28:
Doesn't $21 have to be set to 1 for homing to work?