grbl/grbl #754: About Real-time adjustable feedrates

lilo003, 2015-07-02:
I'm reading in Development Path and Future Needs (https://github.com/grbl/grbl/wiki/Development-Path-and-Future-Needs). The implementation Has Been solved at this point, but just needs to be installed and bugs to be worked in October

How can I test this version?

chamnit, 2015-07-02:
I'll announce a test version when it's ready to be released for testing.

closed 2015-07-02 by chamnit

lilo003, 2015-11-18:
hi @chamnit!

please, some information about Real-time adjustable feed rates?

chamnit, 2015-11-18:
@lilo003 : It will be out very soon! I had a very busy last few months with my day job. I'm finally in the clear as of today.

lilo003, 2015-11-18:
ok, I'm waiting. 

thank you

lilo003, 2015-12-21:
@chamnit, good Morning.

Excuse me again the question, but how is the Real-time adjustable feedrates development?

I am wondering why in Brazil our company developed an interface to control GRBL made to run in the cloud.

And I believe that your firmware is the right choice!

Thank you

lilo003, 2015-12-21:
If you are interested in knowing, be happy.

The link is https://chrome.google.com/webstore/detail/c37-controlador/iodomfcbclgacbhdjmpjfcgdjecdkdhn

And we are preparing to improve the errors that are finding, however, it is already functional

Thank you

chamnit, 2015-12-21:
@lilo003 : Feed and rapids overrides are complete and are working great. But I will not be releasing the source code publicly until I'm closer to a full release. There are multiple reason for this that I won't get into.