grbl/grbl #756: About ARC Interpolation little proposal

blunsmith, 2015-07-05:
I made a CNC controller, I use TI's MSP430F5438A microprocessor. Closer look at your code. Gave me some inspiration, thank you very much, but I do not want to do it from ARDUINO MSP430 transplant. So write their own a controller, I saw one of your interpolation algorithm ARC, personal feel not very perfect, I suggest you use the integral form of the algorithm is even better, so you can get higher accuracy. For the linear interpolation part, because my English is very poor, so it does not look too clear.
Because I want to be a six-axis machine, so I think this linear interpolation control is too important for me. I also used linear interpolation integral form. Integral, behind me have time to explain in great detail. Integral thought I borrowed many graphics functions in the GUI displayed.
Best wishes

chamnit, 2015-07-05:
@blunsmith : I'm sorry but I'm having a very hard time understand what you are saying. The arc implementation is perfectly fine and was done to compensate for how slow trigonometric calculations are on an 8-bit AVR. If you do the math and look at the error, it'll never be less accurate than the physical accuracy of a CNC machine. In other words, calculating a greater accuracy for the sin() and cos() calculations won't improve the arc interpolation.

closed 2015-07-05 by chamnit

blunsmith, 2015-07-08:
I'm sorry, I did not express clearly。 this is my  mc_line() a part。
mcu: MSP430F5438A  16bit   
soft ：IAR 5.30

//采用积分方式的六轴联动方案程序
void step_line(unsigned long* target,float feed_rate)  
{  
    unsigned long xerr=0,yerr=0,zerr=0,aerr=0,berr=0,cerr=0,distance; 
    unsigned long container;//积分所用到的容器

     distance=(long)fmax(fmax(fmax(target[X_AXIS],target[Y_AXIS]),target[Z_AXIS]),fmax(fmax(target[X_AXIS],target[Y_AXIS]),target[C_AXIS]));  //选取基本增量坐标轴 
     container=distance;       //选用最大值作为容器，容器值可以更大,但要求更高的处理器速度。
          
     for(long t=0;t<=distance;t++)  //画线输出  第一次进入循环的时
	{//积分累加
        xerr+=target[X_AXIS];  
        yerr+=target[Y_AXIS];
        zerr+=target[Z_AXIS];
        aerr+=target[A_AXIS];
        berr+=target[B_AXIS];
        cerr+=target[C_AXIS];
        delay_ms(1);
        serial_send_coord(gc_state.position_mm);//发送当前坐标值
      	if(xerr>container)  //积分值判断输出
        {   
            xerr-=container;   
            if(Pin_PUL_OUT & X_AXIS_PUL) { Pin_PUL_OUT &= ~X_AXIS_PUL;} else {Pin_PUL_OUT |= X_AXIS_PUL;}
            //计算当前坐标值
            gc_state.position[X_AXIS] += motor_deriction[X_AXIS];
            gc_state.position_mm[X_AXIS] = gc_state.position[X_AXIS]/steps_per_mm[X_AXIS];
        }  
        if(yerr>container)  
        {  
            yerr-=container;  
            if(Pin_PUL_OUT & Y_AXIS_PUL) { Pin_PUL_OUT &= ~Y_AXIS_PUL;} else {Pin_PUL_OUT |= Y_AXIS_PUL;}
            gc_state.position[Y_AXIS] += motor_deriction[Y_AXIS];
            gc_state.position_mm[Y_AXIS] = gc_state.position[Y_AXIS]/steps_per_mm[Y_AXIS];
            
        }  
        if(zerr>container)  
        {  
            zerr-=container;  
            if(Pin_PUL_OUT & Z_AXIS_PUL) { Pin_PUL_OUT &= ~Z_AXIS_PUL;} else {Pin_PUL_OUT |= Z_AXIS_PUL;}
            gc_state.position[Z_AXIS] += motor_deriction[Z_AXIS];   
            gc_state.position_mm[Z_AXIS] = gc_state.position[Z_AXIS]/steps_per_mm[Z_AXIS];
        }
        if(aerr>container)  
        {  
            aerr-=container;  
            if(Pin_PUL_OUT & A_AXIS_PUL) { Pin_PUL_OUT &= ~A_AXIS_PUL;} else {Pin_PUL_OUT |= A_AXIS_PUL;}
            gc_state.position[A_AXIS] += motor_deriction[A_AXIS];
            gc_state.position_mm[A_AXIS] = gc_state.position[A_AXIS]/steps_per_mm[A_AXIS];
        }
        if(berr>container)  
        {  
            berr-=container;  
            if(Pin_PUL_OUT & B_AXIS_PUL) { Pin_PUL_OUT &= ~B_AXIS_PUL;} else {Pin_PUL_OUT |= B_AXIS_PUL;}
            gc_state.position[B_AXIS] += motor_deriction[B_AXIS];
            gc_state.position_mm[B_AXIS] = gc_state.position[B_AXIS]/steps_per_mm[B_AXIS];
        }
        if(cerr>container)  
        {  
            cerr-=container;  
            if(Pin_PUL_OUT & C_AXIS_PUL) { Pin_PUL_OUT &= ~C_AXIS_PUL;} else {Pin_PUL_OUT |= C_AXIS_PUL;}
            gc_state.position[C_AXIS] += motor_deriction[C_AXIS];
            gc_state.position_mm[C_AXIS] = gc_state.position[C_AXIS]/steps_per_mm[C_AXIS];
        }
    }

    write_SegD(gc_state.position_mm);//把当前坐标值存入seg D
#ifdef DEBUG
    delay_ms(feed_rate);
    coordinate_init();    //测式使用
#endif   
}
#endif

I want to upload a picture , show you.   How to upload photo?

chamnit, 2015-07-08:
@blunsmith : It looks like you are talking about the step segment generation algorithm. You're incorrect. It's actually perfect and uses the midpoint rule, which traces the trapezoid profile exactly. What you propose is very computationally expensive. There are trades to every approach, and the Arduino Uno can only do certain calculations fast and divisions are not one of them.