grbl/grbl #757: 5 or 7 Axis

gorbie35, 2015-07-08:
Hi Guys,

I've used grbl to build a 3 axis cnc mill, which works very nicely. I'd like to build a robot arm next. 
What are the chances of running 5 or 7 axis with grbl? Does anyone have suggestions?

Regards Mike

cri-s, 2015-07-08:
I have done 6 axis on Arduino micro, that have 2 additional pins. But
sources are really old.
g as i remember. I have run it with step punto delay only.

chamnit, 2015-07-08:
@gorbie35 : Grbl is written for Cartesian systems, meaning that each axis acts independently of each other dynamically. I've written about this before in earlier closed issues threads. Robotics is completely different beast. You have motions that multiple solutions and axes motions that depend on each other. The math is difficult and complex to solve in real-time. So, in short, no Grbl will not be a very good solution for controlling robotic arm. It'll probably work ok, but not like how you'd expect or want it to.

closed 2015-07-08 by chamnit