grbl/grbl #758: Stess testing GRBL

J-Dunn, 2015-07-08:
I've been making some additions to GRBL to add make it interface with STM L6474 boards, to provide stronger drivers that the typical circa 2A drivers often used. 

This has added about 1.5K to the compiled GRBL object code. 

I'll be driving fairly substantial mechanical hardware so I need to do some through testing of my modified GRBL before trusting it to power the machine. 

Now GEBL is pretty mature and has ahd considerable testing in the field on a wide range of applications. What I need to examine is whether the mods I've added have introduces new issues, in particular in relation to memory usage. I'm thinking mainly of stack and heap allocation. 

Has anyone done any stress testing on GRBL or can provide suggestions of what areas I need look at most closely and what conditions are like to be worst case senarios?

If I was to write my own stress test, what would the worst thing be throw at GRBL?

Thanks for any suggestions.

109JB, 2015-07-09:
I guess I don't understand what you mean when you say you have made modifications to interface with a stepper driver?  Do you mean you had to modify GRBL code, or are you just making a shield to connect the arduinl to the drivers?

 I myself have GRBL running 3 external stepper drives at 48V and 5A (Keling 6050) with no modifications whatsoever. As far as a stress test, there is a "roadrunner" program that is about 4500 lines long and is composed of many straight line segments that may do the trick for you. You can get the code at the following link.

https://github.com/synthetos/TinyG/blob/master/gcode_samples/roadrunner.gcode

Here you can see my machine running the same code (cutting air) on GRBL and on LinuxCNC 2.6.

https://www.youtube.com/watch?v=c1mxS7WI-tg

I actually just came in from my workshop having run my machine for about the past 3 hours making parts for a new machine build. I am using my own GUI interface and although the GUI has a few quirks to fix, it performed well and GRBL never missed a beat.

J-Dunn, 2015-07-09:
Thanks 109JB , I had to modify the code because the L6474  uses SPI for configuration and status information. Kieling drivers look good but are rather costly. The STM chips handle upto 7A peak and about half that RMS. Evaluation boards with arduino pinout cost about 10 euros. 

Road runner may be one thing but sounds likely to stress the hardware more that the software.

J-Dunn, 2015-07-09:
As part the stress testing and reliability question I came across this: 

http://embeddedgurus.com/stack-overflow/2010/04/efficient-c-tip-12-be-wary-of-switch-statements/

The possibility of the way the compiler chooses to effect the switch statement is particularly worrying in a tight embedded implementation. There is also some code saving to be made by using if () statements.  This mostly affects report.c and gcode.c ( not time critical code ) 

It would be a good idea to remove it from critical code in   ISR(SERIAL_RX) and stepper.c

J-Dunn, 2015-07-09:
I've removed the switch in stepper.c , it saves 12b ( and presumably a little execution time ). I also put the RAMP_DECEL  code first to give it highest priority and RAMP_CRUISE last (longest code path)

reordering the constant declarations ensures the first test ==0  

 ```
 #define RAMP_DECEL 0
 #define RAMP_ACCEL 1
 #define RAMP_CRUISE 2
```

J-Dunn, 2015-07-24:
Sonny, I saw a youtube video with GRBL on Shapoko running at 35kHz ( ? ) that finished with a "braid".

Is that the same thing as the braid_cut2d.nc in the test directory? 

I ran that on my setup and it still functions correctly on the bench, with the mods I've done but it is only doing small displacements. I guess the demo was running full steps, if it was the same thing. 

Would that be a good thing to test the impact of my changes on overall performance? 

thx.