grbl/grbl #759: Arduino Nano with Optiboot

chamnit, 2015-07-09:
I got my hands on a couple of Arduino Nanos and have been looking at getting rid of the big 2.0KB Duemilanove boot loader it uses and replacing it with the much smaller 0.5KB Uno Optiboot. This basically means that you'll be able to cram 1.5KB more into the Nano than before. This is important since Grbl is right on the edge of the Duemilanove's 30KB flash limit, where an Uno has 31.5KB.

I've managed to burn Optiboot onto my Nano to free up that extra space. Anyone know of any problems with doing this?

Here's how I did it and I claim no responsibility on what may happen, even though it seems to be completely harmless. 
- Connect the Nano and AVR programmer to my computer, while the AVR programmer is attach to the Nano's ICSP header. (You can also use another Arduino as an AVR programmer.) 
- Fire up the Arduino IDE. 
- Select "ARDUINO UNO" as the board type in the tools->board drop down menu. NOT Arduino Nano.
- Click "Burn Bootloader" in the Tools drop down menu.

This will then burn Optiboot onto the Nano, free up 1.5KB flash space, and will now behave as an Arduino Uno. Make sure that you select Arduino Uno from then on out as the board type.

csdexter, 2015-07-09:
It should be noted the size in question (i.e. 2KiB vs. 512 bytes) is governed by fuse settings and not by the physical size of the uploaded .hex image ;-) This means, for the new smaller image to actually give you any benefit, you must ensure the proper fuses have been programmed so that the AVR itself knows the "special" area is now only 512 bytes.

chamnit, 2015-07-09:
@csdexter : Yes, you're correct that the fuses sets the 2KB and the 0.5KB sizes. But, I'm pretty sure that the "Burn Bootloader" tool sets those fuses correctly before Optiboot is flashed. Also, the Arduino IDE has set flash limits in their "boards.txt" file for the Uno and the Duemilanove/Nano. Not sure, but this suggests that the Arduino IDE will check and prevent you from exceeding this limit, depending on which board you are using.

chamnit, 2015-07-09:
@csdexter :  I checked the fuses, just to make sure. The Arduino IDE (1.6.5) does burn the fuses before flashing the bootloader.

109JB, 2015-07-14:
@chamnit - How is this working out? Have you detected any issues? If not, I am going to do the same.

chamnit, 2015-07-14:
@109JB : So far so good. I've soldered female pin headers into a couple of Nanos and 3d printed cases for it that's about the size of a packet of gum. They have been my dedicated development boards for the last few days, and I've hadn't any noticeable issues yet with it. Although one of them didn't flash well and returned a lot of garbage in the serial terminal. Not sure what happened there, but burning the boot loader again brought it back.

J-Dunn, 2015-07-26:
/usr/share/arduino/hardware/arduino/bootloaders/optiboot

README optiboot.c  etc looks useful

A couple of configureable option: 
```
/* LUDICROUS_SPEED:                                       */
/* 230400 baud :-)                                        */

/* TIMEOUT_MS:                                            */
/* Bootloader timeout period, in milliseconds.            */
/* 500,1000,2000,4000,8000 supported.                     */
```

J-Dunn, 2015-07-26:
"Not sure, but this suggests that the Arduino IDE will check and prevent you from exceeding this limit, depending on which board you are using."

This is needed for reporting the %age of flash used. I'm not sure the IDE is smart enough to prevent downloading an image which is too big for the target. 

I've heard that you can blow out the bootloader if image is too big. I have not done any destruction tests ;)

J-Dunn, 2015-07-26:
Don't see the  0.5KB Uno Optiboot in there 328 image is about1400B. :?

chamnit, 2015-08-28:
Here's an update on the Arduino Nano running with the Uno's Optiboot. I've occasionally had some problems with the flashing routine not working. It would give a timeout and not flash. The solution is to hard reset the Nano immediately after the flashing routine starts. It takes some practice to get the timing right, but it works. So if anyone runs into this problem, this is one solution.

J-Dunn, 2015-08-28:
This sounds like what you have to do on Uno if you cut the link to prevent it rebooting everytime the serial link is reset. IIRC, avrdude toggles one of the serial lines to force a reset and if you cut the link on the UNO you have to do a manual reset at just the right time.  It seems you have to get it right within about half a second. So, as you say, it needs a bit of practice.

benchuk, 2017-04-23:
@chamnit you have saved my hours of sleep with this Thread - after burning the optiboot I was not able to upload sketch to my nano - did not know I had to select the new board as Uno instead of nano... many many thanks :)