grbl/grbl #760: avrdude: stk500_recv(): programmer is not responding Problem uploading to board.

RichPiet, 2015-07-10:
Hi. I have tried re loading GRBL and re-downloading and then uploading but I keep getting the same error. I have tried new versions of Arduino ide. I now use 1.6.1. I can load and compile ANY example sketch up to my Uno without any errors. I just get this problem with GRBL.

avrdude: stk500_recv(): programmer is not responding
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

stwe12, 2015-07-14:
Hello,
you removed the previous version?

chamnit, 2015-07-14:
@RichPiet : Make sure you select the correct Arduino board type in the tools drop down menu. If that doesn't work, I've found that re-burning the boot loader fixes weird problems like this. Just search the web for Arduino as ISP to use another Arduino to burn it.

RichPiet, 2015-07-14:
I borrowed a friends Uno R3 and that works so I'm assuming the boot loader may be out of date. I will try to re-burn the boot loader. Thanks for the advise.

parallax68, 2015-07-21:
Same problem here, with Arduino Uno SMD, could upload hex file until the v0_9g, after that got the same message. I already tried from several computers, with two different Unos, from different IDE versions on different OS linux and windows.

Let me please give some details here, same output from linux and windows it's not the issue, with the current latest version of the IDE v1.6.5, and latest version of grubl 9.0j, in verbose output for compile and upload.

I'm good for the compilation:

``` cpp
....blablabla...
Sketch uses 28,790 bytes (89%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,485 bytes (72%) of dynamic memory, leaving 563 bytes for local variables. Maximum is 2,048 bytes.
```

And for the upload:

````cpp
avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/blabla/prog/arduino-1.6.5/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/blabla/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 3.3
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "/tmp/build8057143610859504204.tmp/grblUpload.cpp.hex"
avrdude: writing flash (28790 bytes):

Writing | #################################################
avrdude: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x0d
# | 100% 5.79s

avrdude: stk500_cmd(): programmer is out of sync
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
````

It always fail just before ending the upload with this protocol error.

Did someone already gave a try of what chamnit said about re burning the bootloader and could upload with success afterwards?

Also did you dev guys marked the versions in git so I can compare and check if this is a size or feature issue ?

thanks for this great project
brgs
Seb

parallax68, 2015-07-21:
I got it ro work on an Arduino Pro Mini, also ATmega328p, the baud rate is 57600, the device is seen as USB0 instead of ttyACM0, voltage and other little things differ, other values seem to be the same maybe transfer soft and hardware is the point, here is the output for the record:

```cpp
avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/smarty/prog/arduino-1.6.5/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/smarty/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "/tmp/build8057143610859504204.tmp/grblUpload.cpp.hex"
avrdude: writing flash (28790 bytes):

Writing | ################################################## | 100% 9.15s

avrdude: 28790 bytes of flash written
avrdude: verifying flash memory against /tmp/build8057143610859504204.tmp/grblUpload.cpp.hex:
avrdude: load data flash data from input file /tmp/build8057143610859504204.tmp/grblUpload.cpp.hex:
avrdude: input file /tmp/build8057143610859504204.tmp/grblUpload.cpp.hex contains 28790 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 7.18s

avrdude: verifying ...
avrdude: 28790 bytes of flash verified

avrdude done.  Thank you.
``` 
Issue maybe not related to the grbl project but to the transport layer between the ide and the ATmega328p, if someone has an explanation it would be more than welcome.

bits and bytes ;-)

parallax68, 2015-07-23:
Hi again,

problem solved after doing what chamnit said, to burn the bootloader, the burned version is not the same and now I could unload the latest grbl version to my Arduino Uno SMD.

here is the output for the record, note that now the firmware version is 4.4 instead of 3.3 :

```cpp

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "/tmp/build5100055099816542061.tmp/grblUpload.cpp.hex"
avrdude: writing flash (28790 bytes):

Writing | ################################################## | 100% 5.35s

avrdude: 28790 bytes of flash written
avrdude: verifying flash memory against /tmp/build5100055099816542061.tmp/grblUpload.cpp.hex:
avrdude: load data flash data from input file /tmp/build5100055099816542061.tmp/grblUpload.cpp.hex:
avrdude: input file /tmp/build5100055099816542061.tmp/grblUpload.cpp.hex contains 28790 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.15s

avrdude: verifying ...
avrdude: 28790 bytes of flash verified

avrdude done.  Thank you.
```

closed 2015-08-28 by chamnit