grbl/grbl #761: Jog controller

logical, 2015-07-11:
Hi ,

First thank you for GRBL. I had an idea and I was trying to modify the code, unsuccessfully so far. I just thought I would share the idea. I noticed jogging was on the to do list. To control jogging I want to connect an infrared receiver to the unused A4 pin and use an infrared remote to jog the CNC. It might sound like an unprofessional setup but it would be much easier than building a customized panel full of buttons. I  tried to set a pin interrupt and have the interrupt call a function to read the infrared code but the timing of the program is a little difficult to follow.  

Thanks

closed 2015-07-12 by logical

logical, 2015-07-18:
I closed this issue because It was really already discussed and the conclusion was to use the uart to send commands to grbl,so that's what I did. I used a separate arduino to receive infrared and send gcode commands. I think the interface could be expanded by making it programmable  via the receiver and adding an lcd etc.. for example x+100 and enter key would move the x axis 100.

Anyway, thanks again for a great project!

winder, 2015-07-18:
You might be interested in this project:

http://www.shapeoko.com/forum/viewtopic.php?f=11&t=1904&sid=5f2d76172b680c822fcbba37f045bc65