grbl/grbl #762: How much free time in GRBL?

J-Dunn, 2015-07-12:
There are comments in the code indicating that the core ISR must complete within 33us and current worst case is about 25us. That would imply unused processor time of at least 25% ( 8/33 ) on worst case load.

Allowing 1us for overhead that means that there should be slots of about 7us between servicing TIMER1_COMPA ISR, available for doing other things , like occasionally communicating with devices via SPI. 

So, if 50 us is needed for communication over SPI, it will take at most 7 TIMER1 intervals to complete.

Is that a reasonable estimation of the amount of processor time available?

thanks.

langwadt, 2015-07-12:
I'm not sure you can put it that way, the stepper interrupt obviously need to be fast enough to keep up with the fastest step rate, but you also need processing time to interpret gcode and do the planning of what the stepper interrupt should do.

if you are still talking about the L6474, you really don't need to talk to it other than for setup or in the event of an error which is signaled with the flag pin, might even be possible to share the flag pin with one of the limit pins

chamnit, 2015-07-13:
@J-Dunn : That ISR time note references the max 30kHz supported step frequency. If your expected max step frequency is lower, like at 5-10kHz, which is common, you'll be fine.

closed 2015-07-13 by chamnit

J-Dunn, 2015-07-13:
chamnit, thanks, that clarifies. I wanted to see how much can be done without degrading the spec..

@langwadt , yes, it is the error processing which is critical on L6474. Unless you're just going to do an emergency stop everytime something happens, you need to do a fair bit of communication with GRBL running. The best a Uno can do with a stack of three L6474 boards ( with interrupts disabled ) is about 20us the get the status word off one board.

You need to do that three times to find out which one(s) raised the flag and why and which if any have gone HiZ and are no longer driving anything. 

There are some conditions, like thermal warning, where a soft-stop, or other remedial action, could be done without losing counters and without messing up the work peice. Other conditions, like one driver going into HiZ state requires an emergency stop before GRBL does enough steps on the other axes to make a mess of the work. 

These chips provide a useful level of diagnisics. I'm trying to see to what extent this can be exploited in GRBL, though it's probalby pushing the limits of Uno, this is also preparatory work for porting to STM32f4 where it can be handled fully.

J-Dunn, 2015-07-13:
My current strategy on this is to just set a global variable in the ISR that responds to the FLAG pin and get out of the way a.s.a.p. , then initiate the SPI dialogue without interrupts, to let GRBL do it's gerbling. 

If it's a hard error condition, I'll treat it like hard limit pin : set alarm state etc. It's probably going to take about 50 us of non GRBL time to reach that decision point, which is why it can't be done in the ISR . According to the above that is about 7 steps in worst case at top speed. Since these are probalby micro-steps, it should be less than one full motor step. About 25 microns linear movement on a 5mm lead screw. 

I was thinking of servicing the non interrupt part of this at the top of the character parsing loop that reads incoming Gcode. Is that a reliable approach or are there situations in which that is no longer being read? 

thx.

J-Dunn, 2015-07-13:
@langwadt , BTW, how long does it take your bit-bang code to get the status off a daisy-chain of three L6474s ?

langwadt, 2015-07-13:
don't  remember exactly I think roughly 35us. Afaict the datasheet limit for 3*L6474 status read would be ~17us

J-Dunn, 2015-07-14:
Thanks. I've got it down to 18us with a bit of loop overhead and cli() for timing purposes to stop grbl jumping in. I'm marking time with NOPs instead of waiting for the SPIF flag. That improves it quite a bit. 
Doing

J-Dunn, 2015-07-17:
Hell, I'm making this worse than it really is. The circa 19us blast can return all three status words in one hit. So if GRBL has about 8us/33us dead time at 30kHz, dropping to 20kHz means 25us/50us comfortably enough to do both the SPI and service the main GRBL ISR. Worst case, it will cause 20us jitter in GRBL timing on the relatively in frequent, non critical error flag conditions. 

No need to set a global flag and handle it later, it can all be done in the ISR.

I'd still prefer to have the task priorities in the right order but using INT0 should work fine and only slightly degrade the max GRBL speed. 

Thanks for your comments, this looks close to completion.