grbl/grbl #763: Cannot compile and generate .hex file.

yashunandansureka, 2015-07-15:
I downloaded GRBL and am trying to upload it to my Gen7 v 1.5 board (reprap). I made the make file and tried to build the main.c but build errors occur.

Linking: main.elf
avr-gcc -mmcu=atmega1284p -I. -gdwarf-2 -DF_CPU=16000000UL -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -Wall -Wstrict-prototypes -Wa,-adhlns=main.o  -std=gnu99 -MMD -MP -MF .dep/main.elf.d main.o --output main.elf -Wl,-Map=main.map,--cref     -lm
main.o: In function `main':
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:32: undefined reference to `serial_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:33: undefined reference to `settings_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:34: undefined reference to `stepper_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:35: undefined reference to `system_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:49: undefined reference to `settings'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:65: undefined reference to `serial_reset_read_buffer'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:66: undefined reference to `gc_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:67: undefined reference to `spindle_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:68: undefined reference to `coolant_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:69: undefined reference to `limits_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:70: undefined reference to `probe_init'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:71: undefined reference to `plan_reset'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:72: undefined reference to `st_reset'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:75: undefined reference to `plan_sync_position'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:76: undefined reference to `gc_sync_position'
F:\Projects\3d Printer\Firmware\grbl-master\grbl-master\grbl/main.c:85: undefined reference to `protocol_main_loop'
make.exe: *** [main.elf] Error 1

Can someone tell me what the problem is?

chamnit, 2015-07-15:
@yashunandansureka : Grbl only supports the Arduino Uno (and 328p variants). It's not Marlin, nor is it compatible with Marlin/3D printer boards.

closed 2015-07-15 by chamnit

yashunandansureka, 2015-07-15:
okay. I am trying to develop my own 3d printer and i am interested in making  my own firmware. Can you help me out in the part where we send g-codes serially and then decode them?

chamnit, 2015-07-15:
@yashunandansureka : It sounds like you're a bit in over your head.

109JB, 2015-07-15:
The following link will tell you which common reprap firmware will work with your atmega Gen 7 board.

http://reprap.org/wiki/List_of_Firmware

yashunandansureka, 2015-07-16:
Okay. Thanks @chamnit . I will still try though.
@109JB Thanks for the link. I have seen all these before but i was looking for something more simpler, simpler to understand and eventually modify.

Thanks all.