grbl/grbl #764: Cannot deactivate "hold" on steppers

bluebriant, 2015-07-15:
I am using GRBL on a 3 axis CNC system. So far I am impressed with the performance and ease of use. I have 3 Nema 17 type steppers driving the mechanics. I am using 3 Pololu drivers. I have run a check and the accuracy is near perfect. The problem I have is that my steppers stay in hold mode when the power is on and my steppers start running hot. Is there a remedy for this?

chamnit, 2015-07-15:
@bluebriant : If you set the $2 step idle delay to less than 255 then Grbl should disable the stepper motors during a hold.

bluebriant, 2015-07-15:
Thanks I will try this.

bluebriant, 2015-07-15:
OK I tried $2=100, 254 & 0. No luck. The steppers remain on hold, I.e. When I try and rotate them by hand, it is very difficult. I also moved all three axis randomly by jogging them and then tried to hand turn - still no luck.

chamnit, 2015-07-15:
@bluebriant : Check the voltage on pin 8, the stepper enable pin. If it changes when it's moving and when it stops, then there is a wiring problem.

bluebriant, 2015-07-15:
The voltage on pin 8 is 4.74 . When any of the axis are moved the
voltage does indeed change ( to zero). So I guess the wiring requires
checking. My understanding thus far is that pin 8 enables or disables the
steppers. My pin 8 is floating I.e. not connected to anything. Is this
correct? I am not too sure where to start as I have checked the wiring to
the Pololu very carefully (essentially only 3 wires - step, direction and
ground).

The A4988 chip on the Pololu has a SLP (SLEEP) pin, which I assume if
pulled high would disable the power to the stepper. Could this be the
answer? I am reluctant to experiment with this in case it blows the chip.
The data sheet has this to say:

  Sleep Mode ( ̄S ̄ ̄L ̄ ̄E ̄ ̄E ̄ ̄P ̄ ). To minimize power consumption
when the motor is not in use, this input disables much of the internal
circuitry including the output FETs, current regulator, and charge pump. A
logic low on the S ̄ ̄L ̄ ̄E ̄ ̄E ̄ ̄P ̄ pin puts the A4988 into Sleep
mode. A logic high allows normal operation, as well as start-up (at which
time the A4988 drives the motor to the Home microstep position). When
emerging from Sleep mode, in order to allow the charge pump to stabilize,
provide a delay of 1 ms before issuing a Step command.

 The problem now is that on start up GRBL pulls pin 8 to high. Can this be
reversed?

109JB, 2015-07-15:
I think you should read the wiki on configuring GRBL and also the datasheet on the A4988, as all of this is in those documents. Without anything hooked to the enable pin there is no possible way that anything can be disabled or enabled through GRBL.

Regarding the A4988 pins, yes there is a sleep pin, but there is also an enable pin. This is the pin you should be using. Here is the info from the A4988 datasheet.

Enable Input  This input turns on or off all of the
FET outputs. When set to a logic high, the outputs are disabled.
When set to a logic low, the internal control enables the outputs
as required. The translator inputs STEP, DIR, and MSx, as well as
the internal sequencing logic, all remain active, independent of the
 input state.

The GRBL wiki explains the settings. $4=1 will invert the enable pin.

bluebriant, 2015-07-16:
Thank you very much for your help. This has solved my problem.

closed 2015-08-28 by chamnit