grbl/grbl #767: Active state alarm report

Raytonian, 2015-07-19:
I am using Arduino Uno, loaded with GRBL v 8.0 and Universal Gcodesender v 1.0.7 on a 4axis 
ACM Arduino CNC GRBL shield on W7. 
At present I am only driving two small steppers taken from a dot matrix printer and a scanner, on an experimental XY plotter. The exercise is to consolidate the operation of the hard and software in principle for development of a bigger project later.
I am only running sample projects which worked well up to a point, after which everything stopped and I received the report: 'Active state alarm' which I can not reset.
I've read a similar post where the alarm appears during execution of code, but mine is different.
I can reset everything, remove power, plug in, restart Universal Code Sender, with or without gcode loaded and with or without motors connected but the minute I press 'Open', I get the alarm after about two seconds which tells me that, whatever is wrong, prevails.
With the alarm condition present, I can do absolutely nothing and it worked just fine!.
Please help.

Raytonian, 2015-07-19:
Something changed. I re-flashed with GRBL 9, get the same red alarm response but I am now able to reset with $X, but as soon as I enter moving commands, I get a yellow 'Active state: Queue' response and no movement, even with my AZL Ramps stepper drivers removed.

chamnit, 2015-07-19:
@Raytonian : Pins 11 and 12 for z-limit and spindle enable have switched in v0.9, as the disclaimer on the front page states. This would explain your problem. If you can't re-wire your board (hard wired), you can comment out the VARIABLE_SPINDLE define in config.h, recompile, and re-flash to revert to the original pin diagram.

Raytonian, 2015-07-19:
Thanks, Will re flashing with 8 also solve the problem? I only need to prove to some folks that with my contraption that I will be able to reproduce 2d draings at this stage. I guess that by the time you've answered I will have tried it anyway.

chamnit, 2015-07-19:
@Raytonian : Probably. But if you managed to install v0.9 through the Arduino IDE, it'll be a quick fix/check to disable VARIABLE_SPINDLE. Just navigate to the "My Documents/Arduino/" folder (windows) or similar for mac and linux. Open config.h in an editor. You'll see a line `#define VARIABLE_SPINDLE` place a `//` comment in front of it and save it. Then use the Grbl example to upload this new change.

Raytonian, 2015-07-19:
Thanks, I'll do that.

Raytonian, 2015-07-20:
The Arduino folder in My documents does not contain any file like that. Only a 'Lib' folder with one txt document inside. I've activated 'show hidden and system files'.

Raytonian, 2015-07-20:
I think that is because I've used Xloader to flash GRBL to my Arduino and that means I have to get hold of a pre compiled version of GRBL so that I can do the editing before turning  it into a hex file for flashing? I'm learning, I hope.

Raytonian, 2015-07-20:
I have in the mean time reflashed with v 8 and everything stays the same. I still get the red active state   alarm, reset it with $X end after that I get the yellow active state queue response.
I'm pretty much stuck here and I'm running out of options.

chamnit, 2015-07-20:
@Raytonian : A few questions and requests: Can you provide the Grbl `$$` settings that you are using? And can you describe your wiring? And does Grbl work with everything is disconnected from the Arduino, including the shield?

Raytonian, 2015-07-20:
Hope you can make something of this. There is no actual wiring except for the motor leads, which are disconnected at present. I get the active state alarm even without the shield connected. The TX and RX on arduino are contantly flashing and the verbose output below.

**** Connected to COM16 @ 9600 baud ****

Grbl 0.8c ['$' for help]
['$H'|'$X' to unlock]
>>> $X
[Caution: Unlocked]
ok
>>> $$
$0=2000. (x, step/mm)
$1=4000. (y, step/mm)
$2=100. (z, step/mm)
$3=25 (step pulse, usec)
$4=0. (default feed, mm/min)
$5=0. (default seek, mm/min)
$6=130 (step port invert mask, int:10000010)
$7=117 (step idle delay, msec)
$8=-0. (acceleration, mm/sec^2)
$9=-0. (junction deviation, mm)
$10=-0. (arc, mm/segment)
$11=25 (n-arc correction, int)
$12=0 (n-decimals, int)
$13=1 (report inches, bool)
$14=0 (auto start, bool)
$15=1 (invert step enable, bool)
$16=0 (hard limits, bool)
$17=1 (homing cycle, bool)
$18=0 (homing dir invert mask, int:00000000)
$19=0. (homing feed, mm/min)
$20=0. (homing seek, mm/min)
$21=33305 (homing debounce, msec)
$22=-0. (homing pull-off, mm)
ok
[image]

When I ask for verbose input, this keeps running:
ok
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>

chamnit, 2015-07-20:
@Raytonian : Grbl starts up in an ALARM mode whenever you have homing enabled, which you do. You can exit alarm mode with a $X command. 

It also looks like some of your EEPROM is garbled and corrupt. Some of the values don't make sense. This is likely the source of your problems with both v0.9 and v0.8. Please upload v0.9 to your Arduino. See the Wiki to do it through the Arduino IDE (recommended). The newest build of Grbl has an EEPROM wipe command. Simply send Grbl a `$RST=*` when connected to it. It'll wipe and restore the EEPROM defaults.

Raytonian, 2015-07-20:
The verbose input doesn't display right on my post. I actually is:
[verbose]<Idle,MPos:0.,0.,0.,WPos:0.,0.,0.>

chamnit, 2015-07-20:
@Raytonian : That is a problem with UGS not Grbl. You'll have to post an issues thread on their website.

Raytonian, 2015-07-20:
What is UGS?

chamnit, 2015-07-20:
@Raytonian : UGS (Universal G-code Sender)  is the GUI program you are using. It communicates to Grbl, which is on your Arduino. These are two separate programs.

Raytonian, 2015-07-20:
Is there anything else reliable available?

winder, 2015-07-20:
I'm not sure what the issue @chamnit is referring to though. But regardless, teh 1.0.7 version of UGS is over a year old, you should update to 1.0.8 or 2.0.

Raytonian, 2015-07-21:
Ok thanks, I'll try getting a newer version. I'm just not sure  about the
process of flashing GRBL with the Arduino IDE but I'd like to become able
to do that and I guess I'll have to do some reading first.