grbl/grbl #768: Not an issue just throwing appreciation

Mhawley1, 2015-07-22:
Been experimenting with this software for a month now and I am very impressed . Got it to seamlessly work with raspberry pi and am integrating it into a full Bridgeport mill . Bought some motors from ebay to do the job , http://www.ebay.com/itm/4Axis-Nema34-Stepper-Motor-1600oz-in-Driver-DM860A-Power-CNC-Router-Mill-LONGS-/151360747673?pt=LH_DefaultDomain_0&hash=item233dcdb899  . Every time I mention the software everyone says " does it support tapping " to which I grievously say no . Have one video showing it works , ( least for one axis ) https://www.youtube.com/watch?v=SuGkhenp2qE  . 

Some people have complained about the inaccuracy of steppers but I think there is a servo system that takes step and direction . I will experiment and keep at it . My biggest regret is that i don't know enough about QT creator or whichever code was used on the arduino EPROM . Otherwise I would be trying to figure out the rigid tapping problem . Anyways I'm an internet idiot but I have to say you guys are great , keep at it .

csdexter, 2015-07-22:
If by tapping you meant "rigid tapping", that requires a spindle encoder and the processing power to be able to close a servo loop in software fast enough to be useful for tapping. The current Grbl platform (Atmel ATmega328P) doesn't have enough GPIO to accommodate the encoder, enough flash left to accommodate the extra code and enough time left to accommodate the extra calculations.

I think everybody which mentions the "inaccuracy" of steppers doesn't actually understand anything about the issue. Every 200-step-per-revolution stepper ever made has exactly 200 electrical steps to one complete rotation -- no more, no less. The issue of precision doesn't even come into play because the motor is designed and built like that so there's no way you could get 198 or 204 (or some other random value) steps per revolution.
Maybe those people refer to the lack of feedback (compared to professional servo drives) and call that "inaccuracy" which is wrong. A stepper-only system will perform more than adequately if the mechanical forces are always kept within the motor's reach -- that simple.

A servo drive allows you not to care (but would you really want to do that?) because it continuously tries to self-correct -- but if you ram the tool into something too hard for the machine, not even a servo drive will help you cut that piece or achieve the intended precision if the machine is not stiff enough and/or the motors are not powerful enough.

Qt Creator is a toolkit for computers (i.e. desktops, laptops or current-generation phones and tablets) while Arduino is not a computer platform. Grbl is written in plain C and there is no particular IDE you should be using to edit (or that was used to write) the code.

AlexHolden, 2015-07-22:
I think Mathew's reference to QT comes from the fact that he's using Grbl Controller as his sender program.

109JB, 2015-07-23:
As mentioned, steppers are fine and very accurate within their limitations. The benefits of a servo system isn't really accuracy. Servos can be more "accurate", but generally the positioning accuracy of a stepper isn't going to be the determining factor in a machines overall accuracy. 

An open loop stepper system relies on the stepper not being overloaded to maintain positioning accuracy. If it is overloaded, the stepper can "lose" steps. This will only happen if the stepper is overloaded, so it is important to maintain torque loads below what the stepper can handle. In my opinion, the problem here is that if the stepper loses steps, the control software has no way of knowing it and keeps going. As long as you keep the torque requirements within the steppers capability the "accuracy" won't be an issue.

A servo on the other hand, being closed loop, has error checking as a necessity of operation. The popular Gecko Servo drives close the loop at the servo drive. If the actual motor position differs from the commanded position by more than a preset tolearance, the drive can send an error signal back to the control software, letting it know a position error has occurred.  This can be a definite benefit in terms of machine confidence. 

To answer on of the original questions,  there are servo drives that do use step/direction inputs and close the loop internally. The Gecko drives are one example, but there are many more.

In regard to rigid tapping, when someone asks if it can rigid tap, ask if a manual machine can rigid tap? GRBL can't do rigid tapping, and is not the correct choice if that is what you need.

closed 2015-08-28 by chamnit