grbl/grbl #770: exclude Z from soft limits by config

Helicon007, 2015-07-23:
Hello,

I'm using grbl for a PCB milling machine (Cyclone). By design I don't have end stop switches in Z direction, so I perform the homing cycle only in X/Y negative direction. Homing is working very well, but because I don't have homing for the Z axis I'm not able to use the soft limit feature. Zero in Z is more or less the surface of the PCB and I have to move in positive direction for safe move and in negative for milling. With active soft limit the alarm is triggered if I try to move up >0.

Is it possible to exclude the Z axis from the soft limit feature either by configuration or automatically if no homing cycle was done? A gcode to set the Z workspace manually would do the job as well, did I missed it? I'm not sure if I can do this with G54, it seems like I'm on the wrong path to solve my issue.

Regards, Helicon

109JB, 2015-07-24:
I'm curious. Why not add a $1 microswitch to the Z axis for homing. It seems to me that even for pcb work, homing the Z would be beneficial.

Also, you say that zero for z is the pcb surface which makes sense, but that doesn't have to be machine zero. If you zero the work coordinate (G54-G59), then you will be able to move in +Z without triggering an alarm. GRBL always uses work coordinates for programmed moves. The work coordinate system can be offset from the machine coordinate system. For example, I looke up your machine and it looks like it has 3 cm of Z travel.  You can set it up so that with Z all the way up, you are at Z=0 and with Z all the way down you are Z=-3 cm in terms of machine coordinates, but the work coordinate system can have Z up = +3 cm and Z dn=0cm, or Z up=+1.5 and Z dn = -1.5, or any other combination. That way you can have a programmed move to a +Z dimension that is still in negative machine space.

Helicon007, 2015-07-25:
Thank's for your answer, but I was aware of this. Adding a Z switch would solve my problem but the mill has no attachment for a switch by design.

https://github.com/carlosgs/Cyclone-PCB-Factory

Anyhow, it is not impossible to design something. It would just take much more effort for the benefit of soft stops.The mill has no fixed reference point in positive direction and I don't want to attach some weird construct what triggers on top position even if the reference is at the other side. The whole Z axis is sitting at the threaded rod and X carriage and all parts except the steel rods are moving. 

It would be much easier if I could ignore Z or I could set the machine zero manually with a G code. Anyhow, I'm gonna live with that if I can't find a soft solution till it bugs me hard enough and I assemble a micro switch somewhere for Z, I think.

Regards, Helicon

chamnit, 2015-07-25:
@Helicon : Grbl is a general purpose CNC controller. It tries to support the majority of CNC 3-axis applications. It cannot cover them all. 

That said, grbl is written in a way to be fairly simple to modify for something like you want. (See limits.c for soft limits control) Please try doing this yourself to solve your particular problem, as your machine is not a standard configuration. In practice, it really does not make sense NOT to have a homing switch for the z axis. Since your machine is based on a 3d printed design, it's just as easy to design and print a z-limit mounting point for your machine. 

If you're not sure how to do it, I would suggest asking the machine designers to add this functionality, because it's really a core function. You'll be surprised how much more you can do with a machine when it has fixed reference points for all of the axes. Not just two of three.

langwadt, 2015-07-25:
looks to me like most people have added a z-limit switch
http://reprap.org/wiki/Cyclone_PCB_Factory#Community

109JB, 2015-07-26:
QUOTE: "Anyhow, it is not impossible to design something. It would just take much more effort for the benefit of soft stops.The mill has no fixed reference point in positive direction "

The above is absolutely untrue. I have looked at the design and the Z axis moves on 2 vertical rods. It would be trivial to make a bracket to attach to one of the rods and have the Z-axis carriage trigger the switch. Heck, you could probably just hot glue a microswitch to the rod and have the Z-axis homing switch. Saying it is not possible is just not true.

Helicon007, 2015-07-26:
QUOTE: "Saying it is not possible is just not true."

This is exactly the opposite of what I wrote. I know that I can design some hardware solution, I was just asking if there is an other way. Obviously there is no other path, fine. Topic done.

109JB, 2015-07-26:
i apologize. i did misread your post. I think if you tried the z axis homing you would definitely never go back  Perhaps try the hot glue route, which looks like it would be easy on that machine just to see if you like it. I think you will. If you do a more permanent solution can be implemented later.

closed 2015-07-27 by Helicon007

carlosgs, 2015-09-16:
Oh, I have just stumbled upon this thread.  
I agree that a Z endstop can be useful, but it is not 100% necessary in the case of Cyclone (adding it would not only need more parts, wiring.. but also would make it more tedious to dismantle the Z axis when transporting the machine).  

Here is a modified version of Grbl for Ramps that disables the Z endstop and turns it into a probe instead:  
https://github.com/carlosgs/grblForCyclone (it is a year-old version of GRBL though)  
Hope that helps @Helicon007  

For the Cyclone CNC I had been using and recommending Marlin for a long time - what a mistake! GRBL is simply the way to go. Soo robust. Thank you so much for making it!!

carlosgs, 2015-09-16:
Sorry for spamming. Just realized @Helicon007 does not use a RAMPS shield.
To disable the Z soft endstop in your GRBL copy just change limits.c:318. In function limits_soft_check()
```
for (idx=0; idx<N_AXIS; idx++) {
```
to
```
for (idx=0; idx<N_AXIS-1; idx++) {
```

Oh and sorry for the "dodgyness" of the mod (https://github.com/carlosgs/grblForCyclone/commit/55fe29a81640f2dddb381167e8064d2311916fbb ). I was eager to try GRBL in any form possible without needing to re-wire the machine :angel:

Helicon007, 2015-09-16:
Thanks for the info Carlos. I will look into it later. So far I took the short cut and attached a micro switch but it doesn't look nice in my opinion ;)