grbl/grbl #771: Alarm: Homing Fail

redonebe, 2015-07-24:
When i try homing the Z axis moves up and stops immediately with the error message Alarm: Homing fail. When i move the Z axis within 1mm of the Z limit switch the spindle moves up, hits the limit switch, moves down again and then goes up again to do the pull-off in the wrong direction. The X and Y start to move but they stop immediately with the above error.

I connected the logical analyser to the inputs 9, 10 and 12 on the Arduino and only when the switch is hit the signal changes. X and Y stay low while z is showing the changes from low to high when the switch is hit, then low again when moving back and then high again with the pull-off in the wrong direction.

[image]

These are the settings.

$H
ALARM: Homing fail
Grbl 0.9j ['$' for help]
 $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=1 (dir port invert mask:00000001)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=255 (status report mask:11111111)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=4 (homing dir invert mask:00000100)
$24=25.000 (homing feed, mm/min)
$25=1000.000 (homing seek, mm/min)
$26=100 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=400.000 (z, step/mm)
$110=4000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=1000.000 (z max rate, mm/min)
$120=50.000 (x accel, mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

Any suggestions?

chamnit, 2015-07-25:
@redonebe : Your limit switch is either miswired or you'll need to invert the limit pin setting $5=1. Homing behaves this way when Grbl interprets the switch as triggered at the start of the homing cycle.

closed 2015-07-25 by chamnit

nkavinash, 2020-02-27:
i ve differant issue with my grbl. i ve my own built cnc mill,with limit switches on xyz in NO config,and is in trial mode yet ass new learner. few months back it was responding well upon homming command, 1st Zwill go up hit aswitch and pull of on given dist of 2mm,and then XnY simultanously and hit the switch and pull off at 2mm,
meanwhile i ve replaced the arduino board, reflashed grbl etc but now at homming command onle Zmoves up click and pull off, and Alarm comes homming failed and XnY does not moove,
every help will be highly appreciated..my mc config is Z stationory and XY is a cross table..