grbl/grbl #773: Adding means to clear EEPROM to defaults

J-Dunn, 2015-07-27:
https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions

FAQ gives a couple of work-around type solutions for clearing EEPROM to GRBL defaults defined in the source code. However, since the code is already there for the case of a checksum failure,  for the  few bytes more, it would be much better if it was included. 

I have added a extra 'case'  to do this from a new $E command in system.c :
```
        case 'E' : // clear EEPROM
           if ( line[++char_counter] != 0 ) { 
              return(STATUS_INVALID_STATEMENT); 
           }
             else { settings_init(0); }
           break; 

        case 'H' : // Perform homing cycle [IDLE/ALARM]
          if (bit_istrue(settings.flags,BITFLAG_HOMING_ENABLE)) { 
            // Only perform homing if Grbl is idle or lost.
            mc_homing_cycle(); 
            if (!sys.abort) { system_execute_startup(line); } // Execute startup scripts after successful homing.
          } else { return(STATUS_SETTING_DISABLED); }
          break;
```
It involves adding a parameter to settings_init() and calling it with zero to clear EEPROM; the only other call is from main which now needs a non-zero value (1)

settings.c
```
// Initialize the config subsystem
// option==0 restores GRBL defaults, replacing used defined values in EEPROM
void settings_init(uint8_t option) {
  if ( (option==0) || (!read_global_settings()) ) {
    report_status_message(STATUS_SETTING_READ_FAIL);
```

There may be better suggestions than $E , it may at least be preferable to test for a second char $EE to avoid gobbling  up all letters that may be useful in future.

chamnit, 2015-07-27:
@J-Dunn : Grbl v0.9j has an EEPROM clear and restore to default command. See the Wiki configuring page for details.

closed 2015-07-27 by chamnit

J-Dunn, 2015-07-27:
Thanks,
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9
"These commands are not listed in the main Grbl $ help message, but are available to allow users to restore parts of or all of Grbl's EEPROM data."

Then it may be helpful to add them to GRBL command list, as well as correcting the FAQ which explicitly states that this feature does not exist.  I tired GRBL help and found nothing, so I came here looking to find out whether this was possible and once I'd read that in the FAQ, I stopped reading and started coding it. 

A simple oversight I'm sure.

chamnit, 2015-07-27:
Not an oversight. It was intentional as the wiki page states. Program strings take up a lot of valuable flash space. 

The Faq was updated a few minutes after I posted the last message.

J-Dunn, 2015-07-27:
The oversight comment was in relation to the FAQ. Thanks for fixing it.