grbl/grbl #775: Feed Halt

AlexHolden, 2015-07-30:
I'd like to suggest adding a 'feed halt' realtime command. It would flush the planner queue and serial input buffer and bring all axes to a controlled stop. This would be useful for senders to be able to quickly and cleanly interrupt a stream of jogging moves when the user lets go of the jog button.

I understand it is currently possible to emulate this behaviour by:

Send a feed hold
Poll the status until grbl returns to idle to indicate that the motors have stopped
Send a soft reset to flush the buffers

However this method seems a bit kludgy and inefficient to me.

AlexHolden, 2015-07-30:
I should add that ideally it would be possible after issuing a feed halt command to immediately start sending new blocks that will start executing once the axes have stopped moving, rather than having to poll for the idle state first. (For when the user immediately changes from one axis jog button to another without waiting for the machine to fully stop moving first.)

chamnit, 2015-07-30:
@AlexHolden : There are a few closed issues that discuss this. In short, no. To flush the Grbl buffer can have unintended consequences if someone doesn't fully understand what they are doing. Esepecially if they do it during a job. The solution is have a dedicated continuous jog command that internally avoids having the user flush the buffer. If I have room in v1.0, I plan on installing it.

closed 2015-07-30 by chamnit