grbl/grbl #777: Allow to reuse Z min as probe sensor

Nandox7, 2015-08-02:
Hi,

At the moment the pins for the Z min and the probe are different.
I was trying to find use cases where that makes sense but I can't find many. 

For my use case that is mainly doing PCB's this means that I need to swap the connection of the probe between the two first when homing the Z axis and later when probing the board.

Ideally when the G38 command was issued it would swap the detection from the Z min to the probe.
At the moment it will lock the progress as the Z min is reached.

Hope this makes sense.
Cheers

chamnit, 2015-08-02:
Grbl has no Z min pin, so no, your comments don't make much sense.

Nandox7, 2015-08-02:
What about what is since a few version sin D12? Z min limit?

From the Wiki: "Z-Axis limit input on D11 has swapped with spindle enable D12 to support variable spindle PWM output.:

chamnit, 2015-08-02:
Limit pins are used for homing and hard limits. Homing determines a repeatable machine space. For z, this is usually z+. Not z minus. Probing determines the touch point of an object inside the machine space. 

I see no problem here. Just a misunderstanding of setting up a CNC. So I'm closing this issue.

closed 2015-08-02 by chamnit

Nandox7, 2015-08-03:
Well it was not a problem but a request to expand an existing feature.

I have a mini CNC a Cyclone PCB Factory and it doesnt have a limit switch for the Z axis. When homing the Z axis it works similar to a probe so it determines the limit of the Z when the bit touches the pcb.

Now the reason I asked for this feature is because at the moment because the pin for the Z limit and the probe are different the probe has to be disconnected from one pin and connected to the other.

If there was a flag that would allow the Z limit switch to act as a probe when the G38 code is used it would avoid redundancy of having two probes or to have to reconnect it.

chamnit, 2015-08-03:
@Nandox7 : This is the second time a Cyclone user has requested something similar. Frankly, I don't understand why the Cyclone doesn't have a z homing switch or they don't provide a methodology for you. The way I see it is that you have three options.

- Add a z-limit switch. The Cyclone is 3d printed. Just 3d print a z-limit mount for z+. This is recommended as the z-coordinate system will never change based on the PCB thickness. This allows for much more functionality, such as work coordinate systems, and this is they way a CNC should be run.

- Ignore and disable the z-homing. You can do this in config.h. You can have Grbl only home x and y, and since you are probing, it doesn't matter where z is initially. But I still highly recommend doing option one.

- If you are still adamant about locating at z- with homing, I would touch off on the table, NOT the PCB. The table is a consistent reference point, which can be done by a separate switch or something else. Probing would then touch off on the PCB to locate the surface in machine space. 

Again, I still see no problem here, except that the Cyclone design is flawed and encourages improper use of a CNC machine. Having to home to the PCB and then probe the PCB doesn't make any sense.

Nandox7, 2015-08-03:
Thanks for the insights, despite I'm quit acquainted with 3d Printers CNC's are new to me so it is expected that I use incorrect terminology and concepts.

Your explanation makes sense and I understand now why this that seemed obvious wasn't yet implemented in GRBL.

It was in my plans to improve the way the Z axis work so adding a Z limit switch will be yet another features.

Thanks again for the details shared.