grbl/grbl #778: Arduino IDE's new Library Specification prevents loading grbl as Library

Shanjaq, 2015-08-03:
Please see Arduino's new library.properties specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

Until the library.properties file is created for grbl, any attempt at importing grbl results in the following error:  "Specified folder/zip file does not contain a valid library"

109JB, 2015-08-04:
You don't use the import function of the IDE. You simply copy the contents of the girl folder to your library folder and then start the IDE. Specific directions are in the wiki.

J-Dunn, 2015-08-23:
I think you will find that that error comes from downloading the zip from github which automatically creates a structure with the name grbl-master. 

Now the Arduino pseudo-IDE is too dumb to cope with anything but alpha characters in library names ( yeah, really ) and too dumb to give an intelligent and meaningful error message when it finds something else. Instead of warning about invalid chars  or better still allowing stuff like hyphen and underscores in names it just plays dumb and says there is no valid library.

You need to unzip it, rename the directory grbl-master to  grbl and rezip it. Then the pseudo-IDE will be able to unpack it.  Otherwise use git-clone command.

Then of course in true M$ mentality you will need to close the pseudo-IDE and restart it for it to take the newly installed lib into account, since having  installed it itself it is too dumb to know it is there.

You may have guessed by now that I'm not too impressed with their so-called IDE.  I would recommend using Arduino-Makefile to build and flash GRBL to Arduino and a decent, syntax-aware, code editor like geany to edit the code.

https://github.com/sudar/Arduino-Makefile
https://github.com/geany/geany

109JB, 2015-08-24:
Geany may be great, but for someone just wanting to upload grbl to an arduino, the arduino IDE works fine and the above is way too convoluted. It is really very simple to upload through the IDE

1 - download zip file from github
2 - Unzip.
3 - Within the unzipped download is a folder named "grbL" . Copy that to your Arduino Libraries folder Mine is C:\Users\John\Documents\Arduino\libraries
4 - Done

Now just start the IDE, go to examples/grbl/grbl upload and it works fine

J-Dunn, 2015-08-24:
So are you agreed that this 'bug' is nothing to do with GRBL but is a result of the automatic github zipping process that adds "-master" to the directory name and that the error message is due to the presence of the hyphen?

chamnit, 2015-08-24:
Please follow the Wiki directions and ignore the ".ZIP" of "Add .ZIP Library". You can choose an unzipped folder, just like before. It works fine.

closed 2015-08-24 by chamnit

109JB, 2015-08-25:
@J-Dunn 
As far as I can tell, the hyphen in the folder name has absolutely nothing to do with any problem. I just now renamed the "grbl" folder which is contained in a subfolder of the git zip file to "grbl-master". I then used the IDE to add this folder as a library, and I then uploaded grbl to an Arduino without any problem. The hyphen is still in the folder name, and it comes up as "grbl-master" in the examples pull-down of the IDE.

Also, using the IDE to install the library results in the library being immediately available. It isn't necessary to close the IDE and restart it for it to know the new library is there. 

I would venture a guess that many people who use grbl never need to edit any of the grbl files for their use of grbl. Rather, they just want to upload it. While editing may be limited in the IDE, it still provides a simple way to upload if simply following the instructions in the WIKI.

While you may not like the Arduino IDE, and there may very well be other options better suited for your uses, let us not forget that "Arduino" is a tremendously successful project and the Arduino IDE is FREE software, and the Arduino platforms are FREE designs that have helped countless people build interesting projects (grbl included) while learning about microcontrollers, sensors, programming, circuits, etc.