grbl/grbl #780: End of Line Confusion

AlexHolden, 2015-08-10:
Grbl always sends DOS-style two-character EOL sequences (i.e. '\r\n') in its replies, however it seems to expect to receive a single character EOL at the end of each command (either '\r' or '\n' alone will work). If you send it a command that ends with '\r\n' it interprets the '\n' as a blank line and replies with an extra 'ok'. Is that what's supposed to happen?

JohnKnit, 2015-08-18:
I can't tell if its supposed to happen, but can confirm it does. Maybe that is something to add in the wiki as it only says there grbl answers with 'ok' to every successfully completed line.

chamnit, 2015-08-18:
@AlexHolden : Yes. It's supposed to happen this way. Changing it will likely break a lot of GUIs, as they all expect this behavior.

closed 2015-08-18 by chamnit