grbl/grbl #781: More Than One Error from a Command

AlexHolden, 2015-08-11:
I think I may have found a problem with the character-counting streaming protocol. I'm implementing it by storing each command I've sent in a FIFO and keeping a running total of the number of characters I've sent. When grbl acknowledges a command I pull it back out of the FIFO and subtract its length from the running total. For this to work I have to be able to associate each command sent with an acknowledgement received. grbl can acknowledge a command with either an 'ok' or an 'error:' string. Fine.

The problem comes when grbl generates more than one 'error:' response from a single command, which my sender interprets as two commands having been acknowledged. Am I missing something?

For example if I was to send:
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
I get these two error responses:
error: Line overflow
error: Bad number format

AlexHolden, 2015-08-11:
I wonder if what's happening is grbl is discarding the too-long line at the point where its buffer is full and issuing the first error, then trying to execute the rest of the line as a new command and generating a second error because it is garbage. Perhaps grbl should silently discard the rest of the too-long line instead of trying to execute it?

chamnit, 2015-08-11:
@AlexHolden : Nice catch. I'll fix this on the next push.

AlexHolden, 2015-08-11:
Cool, thanks.

chamnit, 2015-08-28:
Fixed in v1.0 edge branch. This problem will remain in v0.9 so not to risk incompatibilities with current GUIs.

closed 2015-08-28 by chamnit