grbl/grbl #782: Current status doesn't work during homing

vlachoudis, 2015-08-12:
During the homing cycle grbl is not reporting to the current status '?' command
Is it a feature to perform the homing more precisely?

chamnit, 2015-08-12:
@vlachoudis : It's not exactly a feature, but it's a consequence of how the homing cycle is implemented. The main program has to monitor the homing switch in real-time and can't take the time to send a report during it. It's done in this way to avoid creating unnecessary volatile variables, as this creates a whole host of other problems. If it sends a report, the homing cycle monitoring may miss the switch change. 

At some point, I may revisit the homing routine to allow reporting during it, but I didn't find it really necessary, since position is indeterminant during a homing cycle anyways.

vlachoudis, 2015-08-13:
@chamnit indeed I understood like that. Naively I was expecting to see the state "Home" during the homing cycle but since grbl was not reporting anything bCNC was reporting the last known state.

closed 2015-08-13 by vlachoudis