grbl/grbl #784: more appreciation!

splitn2, 2015-08-17:
Hi there, I've just started bench testing a small GAUPS board http://store.amberspyglass.co.uk/gaups-1-0-arduino-compatible-stepper-shield-kit-50v.html and close Uno arduino with GRBL and 3 small nema17 steppers. Totally blown away by this as a very cost effective way to build a small router or mill. I have a larger AC servo drive system on the bench test too as part of a larger build running Mach3 but I have to say so far this GRBL open source cnc with the variety of cheap stepper drives is really impressive. Thanks for all the effort you guys put in to developing this.

I do have a question, what is the most active forum for GRBL discussion? I'm keeping an eye on this Git of course for updates but if there is a more general chat area I'd love to know.

cheers
Richard

splitn2, 2015-08-30:
hi @chamnit , just wondering when you have time is it possible to update the community on the active areas of work on GRBL? Where is it heading what are approx plans for future releases? I read the Wiki but I'm wondering what the latest plans are? No hurry I am just curious and am loving how neat GRBL is so far. All I am doing is bench testing it. I am interested in the Flash card system that can run autonomous that looks very useful. Keep up the great work and thanks for your amazing contribution to the CNC world! Cheers Richard

chamnit, 2015-08-31:
@splitn2 : I've decided not to publicly divulge future features until they are very close to release. There are a number of reason for this, but mainly, by saying anything, I'm committed to doing it. I have plenty of these 'commitments' already. 

That said, Grbl v1.0 is under heavy development at the moment. Most of which has not been pushed to Github. There are some very powerful features that are coming, but it's mostly a question whether or not it will fit onto an Arduino Uno (328p) or not. If it doesn't no big deal, all of the new features will run just fine on an Arduino Mega2560.

As for central discussion or forum for Grbl development, this is it. If you want to chat about applications or troubleshooting, the shapeoko.com Grbl forum is one of the most active with a great community to help you out.

splitn2, 2015-08-31:
awesome I am waiting with anticipation @chamnit  !!!!!! thanks again, you guys are awesome!