grbl/grbl #785: optocoupler and pwm to analog questions

109JB, 2015-08-21:
So this isn't exactly a grbl question but a grbl application question.  I am getting ready to play around using grbl to control a motor, but the motor drive doesn't have pwm inputs. It has a potentiometer that varies voltage from 0 to 10V.  I also want to isolate this from the Arduino.  I'm not very good at electronics, so I'm asking you electronic type guru's to take a look at my drawing to see if I am on the right track. Let me know what you think.

[image]

AlexHolden, 2015-08-21:
I'm too rusty at this stuff to comment on the resistor values, but I would recommend not using a 741 for this application because it isn't rail-to-rail. That is to say, if you supply it with a 12V-0V single rail supply the output can't go all the way down to 0V or up to 12V.

109JB, 2015-08-24:
Thanks @AlexHolden   I saw your response and I think I may go a different way.  The drawing above basically is 3 parts, 1-the optocoupler, 2-A low pass filter, and 3-The op-amp to boost to higher voltage.

I think the way I may go about it is by feeding the 0 to 5V PWM from the arduino to the base of a transistor to switch a 10V source. This should get me 0 to 10V PWM. From there, I can send it through a low pass filter or two to smooth it out a bit. 

I have the basics in my head, but haven't had time to put together a drawing and probably will be a couple days before \i will have a chance since I'm getting sent out of town.

109JB, 2015-08-28:
Got back home and here is my new idea for this.  Here is how I think this will work

1 - The PWM from the Arduino is sent to the base of an NPN transistor while the collector and emitter are connected to the 5V and GND pins.  This will result in a 0-5V PWM signal but the duty cycle will be inverted from the original PWM from the Arduino.

2 - The inverted PWM is sent through a optocoupler to isolate the Arduino from the motor controller side.

3 - The output from the optocoupler (Inverted PWM) is sent through another NPN transistor that has its collector and emitter connected to the 10V and 0V connections on the motor control where the original potentiometer was connected. This will again invert the PWM duty cycle, but will also result in a 10V PWM. The duty cycle should now match the original PWM from the Arduino.

4 - Finally the 0-10V PWM is sent through a low pass filter to convert it to 0-10V Analog

Of course I have some other components in there like load resistors, a regulator for the opto, and a cap for the low pass filter. I did find an online electronics simulator to play with and the circuit appears to work, but  I'm far from any kind of expert. 

Well, that is how I think it will work, but I'm an amateur when it comes to electronics so any help would be appreciated.

Thanks,

John B.

[image]

AlexHolden, 2015-08-28:
Hi John. The original op-amp circuit seemed simpler to me, apart from needing to change to a rail-rail device if you need 0V output. I'd also include a pot in the feedback network so you can tune the amount of amplification, and consider powering it from the 10V rail so you don't need to worry about accidentally overdriving the motor controller.

You don't give a part number for the opto isolator you're using, but the one in your circuit diagrams has an inverting amplifier in it (the little circle on the output means it's a 'not' gate). I don't think the transistor on the microcontroller side is necessary - the AVR GPIOs should be powerful enough to drive the LED directly via a current limiting resistor. If you need to invert the waveform, connect the GPIO to the cathode and 5V to the anode.

I'm guessing the 5V regulator is because the opto has some active logic in it that specifically requires 5V?

What exactly will happen on the output is impossible to predict without knowing the input characteristics of the motor controller. If you're lucky it may be going straight into a high impedance device like an A-D converter. That's why the op-amp in the original circuit was a good idea - it has an active low-impedance output driver that isolates you to a large extent from the characteristics of the circuit you're feeding your signal into.

You're going to find an oscilloscope extremely useful in getting this working. Good luck!

109JB, 2015-08-28:
Thanks Alex.  I do have an old o-scope I can hook up. The pot in the feedback network is a great idea rather than the 2 fixed resistors. 

The opto I was looking at is the 6N137. I see what you are saying. If I use the non-op-amp circuit that I show in the second circuit, then my analog output will have an inverted duty cycle due to the opto. good catch. 

I think I will try the second circuit first as I have all of the parts for it except the opto already. I'll initially breadboard it up with the 2 NPN transistors and no opto like the simulator circuit and see how it goes.

If I do wind up going to the op-amp circuit, would you have a part number recommendation on what op-amp to use? 

I probably don't need the output to go all the way to 0V, since I will be tying this circuit in with the spindle enable pin on the arduino. I can use that to shut off the motor controller completely, but I am not sure if the motor controller needs to "soft start" at 0V before ramping up to set speed. It has been a while since I messed with the actual motor controller I will be using. I guess I will find that out once I get it hooked up. 

Thanks for your help

AlexHolden, 2015-08-31:
Hi John.

It looks like that opto-isolator has an open drain output; that means it can only pull down, not pull up. You can probably get around that by adding an external pull-up resistor. It also looks like you don't have to connect the enable pin (though doing so won't do any harm). If you haven't bought it yet, for this application I would be inclined to instead go for something with a simple output transistor that doesn't need the 5V supply.

 One difficulty these days is finding op-amps that are still available in DIP packages! The ST TS912 looks pretty nice if you can find it. One advantage of it is that it's a dual op-amp, so you might consider a different circuit configuration (not tested - this is straight off the top of my head!):

Feed the signal from the opto isolator into the first op-amp in open loop (i.e. infinite gain) configuration to produce a 0-10V square wave on the output. You can use either the inverting or non-inverting input depending on whether you need to invert the signal.
Feed the output of the first op-amp through an RC network to convert it to an analogue voltage. You might have to experiment with the time constant to get the right balance between ripple and response time.
Feed the analogue signal through the second op-amp in unity gain buffer configuration to get a low output impedance you can send on to the motor controller.