grbl/grbl #786: Created another interface to Grbl with WebSocket

cho45, 2015-08-26:
Hello, I created another implementation of interface to Grbl.

- https://github.com/cho45/GrblServer

This works on node.js (WebSocket server, interface to Grbl directly with serial) and browsers (WebSocket clients, UI).

The server is not just proxy to serial but it manages executing G-code and Grbl states. So its clients are always reloadable on browsers (clients are just UI). It means multiple browser sessions are allowed in same time for controlling from remote, viewing to machine-local display and etc..

chamnit, 2015-08-26:
@cho45 : This looks fantastic! Thanks for sharing! Do you have an idea how well the Raspberry Pi web browser can handle running this locally?

cho45, 2015-08-26:
I tried it but browsers in Raspberry Pi with LCD is too slow to control... (I connected LCD via SPI. HDMI connection may be more good performance) So, I use the LCD to just showing coords and states currently.

bubalazi, 2015-10-13:
Great implementation and looks like a good chance to practice some typescript.