grbl/grbl #787: Homing Cycle Z-Axis not working properly

zorgoban, 2015-08-28:
After updating to the new version I had to change homing pull-off from 1 to 2 mm to work properly at all. No problem so far. Now I often observe wrong behaviour on the Z-Axis. Instead of moving to the limit switch and after that back and forth two times it just moves down two times and starts homing X and Y.

zorgoban, 2015-10-16:
I now found out that this behavior was caused by a power cable running alongside the signal cable. Unfortunately they are not yet shielded. It's still a bit sad no one responded to my issue.

closed 2015-10-16 by zorgoban

chamnit, 2015-10-16:
@zorgoban : Please be conscious that this is an open-source project that is supported by the free time of others. Everyone is doing what they can. 

In addition, if you took a little bit of time and made a slight effort, you would have found out that there have been dozens upon dozens of closed issues similar to yours that have been resolved, several other websites and forums (like shapeoko.com) that have solutions posted, and a Wiki page explaining the problems of electrical interference on a signal line. Please do your homework before going off slandering volunteers.