grbl/grbl #788: Grbl at 20 Mhz

sergiv69, 2015-08-28:
Hi,

Sergi from Barcelona.
I've build my own board based on a 1284p, so far so good. However as Grbl is not arduino dependant, I supose it should be easy to have it running at 20Mhz... Except that it was not that easy...
When I put the 20Mhz crystal, it does strange things. So I think I should need to change something in the code, but what???

langwadt, 2015-08-28:
I'd assume that  changing CLOCK = 16000000 to CLOCK = 20000000 in the makefile is all that is needed

chamnit, 2015-08-28:
I haven't tried this, but I think @langwadt is right. However, you'll need to compile and flash via terminal or some similar method. The Arduino IDE upload method will not work, because it assumes the clock speed is 16MHz. It's hard to change.

sergiv69, 2015-08-28:
Aha! It's the IDE!
Via terminal... Can I use the same avr-gcc that the IDE uses?
Or any sugestions? 
(Will be very welcome!)

timryder, 2015-08-31:
I use always the avrdudeSS program. 
You just need the .hex file which you can easily find if you're compiling in Arduino IDE or better to just download the .hex from here.

J-Dunn, 2015-09-02:
Arduino-Makefile project here on github also used avr-dude,  will compile and flash GRBL and  allows you to define F_CPU

sergiv69, 2015-09-02:
Uhhh, very interesting!!!
Can I also upload it using a programmer like atmel ice or avr?