grbl/grbl #789: Grbl 4th rotary axis

sergiv69, 2015-08-29:
Me again...
I've setup GRBL for X Y Z A, axis. Surprisingly works ok and interpolates all axis. Great!
But, that is only half of the work. Now I can't interpolate properly the rotary with the linear axis. (That was as expected).
For example, when trying to interpolate Y and A (which is really C, as it is perpendicular to Z) this way:
G01 Y5 A5 F20
Instead of having a line drawed, I got and archimedes helix.
Thats normal, as A radius changes when Y moves, so the perimeter changes.

The solution I thought is:
GRBL draws lines as lines, ok. But circles as small lines.
So in order to work in this config, I need to change motion to draw both lines and circles as small arcs (in the example between Y and A).

1.Do you thing this could work or there is a better solution?
2.How can I properly do it in order to have some uses for the community? (I think the line/arc drawer and new line/arc drawer should coexist, so when A axis is not involved the rest of the axis could interpolate correctly)

109JB, 2015-09-01:
@sergiv69 

I am a little confused as to what the problem is. You reference trying to make a linear move when using a XYZC setup, and it sounds like the work is being performed in the XY plane. A linear move in that case would simply be a move in X and Y since the C axis is perpendicular to the Z axis. You would not require moving the C axis to make that move. 

On another note, I am nearing the point where I will be implementing a 4th axis. Can you tell me what modifications to grbl were necessary to incorporate the 4th axis. I would be using a MEGA 2560.

ashelly, 2015-09-02:
@109JB, You can look at https://github.com/grbl/grbl/pull/703 for the changes I made to get a 4th axis working. I am curious to see if sergiv69 came to a similar solution.

EliteEng, 2015-10-23:
If you look at rotary axis as angular and not linear it becomes a lot simpler.
A values should be in degrees.

mkeyno, 2016-04-13:
hi guys, I have same question , actually I decide to engrave image or letter on the metal ring , I've just intend to aligned my penomatic hammer with central line of rotary axis and use only axis A & Y , any help or links really appreciated